anholt/linux #11: vc4: HDMI power on when not connected at boot not supported.

anholt, 2016-02-15:
Made progress on this on 2016-02-15, but not quite there yet.

anholt, 2016-03-11:
Patches are headed upstream for this one: https://lists.freedesktop.org/archives/dri-devel/2016-February/100963.html

schnitzeltony, 2016-03-12:
I've put the series mentioned on top of raspberry Pi 4.1 kernel. I see:

* Kernel boots but it crashes later in userspace when startig X with
[    9.870456] Division by zero in kernel.
[    9.874323] CPU: 2 PID: 140 Comm: systemd-udevd Not tainted 4.1.19 #1
[    9.880755] Hardware name: BCM2709
[    9.884187] [<80017200>] (unwind_backtrace) from [<80013824>]
(show_stack+0x20/0x24)
[    9.891939] [<80013824>] (show_stack) from [<8058c008>]
(dump_stack+0xcc/0x110)
[    9.899254] [<8058c008>] (dump_stack) from [<8001375c>]
(__div0+0x24/0x28)
[    9.906134] [<8001375c>] (__div0) from [<803008a0>] (Ldiv0_64+0x8/0x18)
[    9.912754] [<803008a0>] (Ldiv0_64) from [<8047952c>]
(bcm2835_clock_choose_div+0x30/0x70)
[    9.921019] [<8047952c>] (bcm2835_clock_choose_div) from [<80479588>]
(bcm2835_clock_set_rate+0x1c/0x44)
[    9.930499] [<80479588>] (bcm2835_clock_set_rate) from [<80474ba8>]
(clk_change_rate+0xac/0x3a0)
[    9.939326] [<80474ba8>] (clk_change_rate) from [<80475af8>]
(clk_core_set_rate_nolock+0x78/0xc8)
[    9.948243] [<80475af8>] (clk_core_set_rate_nolock) from [<80475b78>]
(clk_set_rate+0x30/0x48)
[    9.956906] [<80475b78>] (clk_set_rate) from [<7f12fac8>]
(vc4_hdmi_bind+0x18c/0x400 [vc4])
[    9.965308] [<7f12fac8>] (vc4_hdmi_bind [vc4]) from [<8038252c>]
(component_bind_all+0xc0/0x1ec)
[    9.974165] [<8038252c>] (component_bind_all) from [<7f12d248>]
(vc4_drm_bind+0xb8/0x190 [vc4])
[    9.982906] [<7f12d248>] (vc4_drm_bind [vc4]) from [<80382730>]
(try_to_bring_up_master.part.0+0xd8/0x118)
[    9.992566] [<80382730>] (try_to_bring_up_master.part.0) from
[<80382830>] (component_master_add_with_match+0xc0/0x138)
[   10.003373] [<80382830>] (component_master_add_with_match) from
[<7f12d3d0>] (vc4_platform_drm_probe+0xb0/0xbc [vc4])
[   10.014009] [<7f12d3d0>] (vc4_platform_drm_probe [vc4]) from
[<80388dfc>] (platform_drv_probe+0x3c/0x70)
[   10.023493] [<80388dfc>] (platform_drv_probe) from [<80387568>]
(really_probe+0x1c0/0x29c)
[   10.031794] [<80387568>] (really_probe) from [<80387750>]
(__driver_attach+0xa8/0xac)
[   10.039687] [<80387750>] (__driver_attach) from [<803858a0>]
(bus_for_each_dev+0x78/0xac)
G[   10.047879] [<803858a0>] (bus_for_each_dev) from [<80386f88>]
(driver_attach+0x2c/0x30)
[   10.056127] [<80386f88>] (driver_attach) from [<80386c0c>]
(bus_add_driver+0x15c/0x204)

* Starting with HDMI connected and changing video mode causes monitor black
and complaining for invalid timing.

greyltc, 2016-03-13:
awesome! can't wait to test it!

anholt, 2016-04-20:
Patches merged upstream, and the 4.4 and 4.5 pull requests downstream should cover it as well (just tested again on 4.5)

closed 2016-04-20 by anholt

ryancito02, 2016-05-15:
Maybe this is the problem I'm experiencing. 
Just posting so this reference shows up on the issue page.
v                   v                  v
https://github.com/anholt/linux/issues/28