anholt/linux #12: vc4: dmabuf import from camera driver not supported

anholt, 2016-02-15:
We need this for zero copy display of camera contents.

MarcoBodega, 2016-03-15:
+1 on everything related to camera

6by9, 2016-08-02:
Hi Eric.
What do you need from the camera subsystem to support this? Or is this a reminder to yourself that you need to support dmabuf importing?
I presume buffers need to be either allocated from VCSM and the reloc heap, or CMA and passed in to V4L2/MMAL, but whichever way you'll want them contiguous.

I'm hopefully going to be looking very soon at what I can do with VCSM and dmabuf (import and export hopefully), so please shout (or we can take the conversation offline) if there is anything in particular that would help you and I'll see if it's possible.

Are we going to have fun with the ARM and VPU fighting over power and clock controls? Ought to coordinate that with PhilE or Dom. There was code for BCM28145 that did have the VPU talking to a kernel service to deal with clock and power requests. Would that help and make the ARM in charge of everything? There's probably easy access to the VPU side code, but probably not the kernel driver.

  Dave

anholt, 2016-08-02:
I think it should be fine for the camera subsystem to be generating dmabuf buffers.  My expectation is that I just need to have vc4 walk the page list, make sure they're contiguous, map the paddr to kernel virutal, and set up the drm_gem_cma_object's paddr/vaddr appropriately.  I also need to then have vc4 participate in fencing, which I haven't looked into too much yet.

I would *love* to have the VPU be asking the linux kernel to do power and clock management (we only have a clock driver so far, but it would be great to control power, too).  I suspect that will be a challenging architecture change to get Dom/Phil on board with, though.

6by9, 2016-08-02:
So currently the V4L2 driver produces non-contiguous buffers as VCHI is doing a copy from VC reloc heap (contiguous) to ARM buffers.
There is a zero copy option that will expose the reloc heap handle from which you can get the phys addr fairly easily. There are a couple of examples kicking around of userspace using MMAL and zero copy, but not mainstream.
Fitting that in with V4L2 is not trivial, however I'm working on the reverse of getting V4L2 to use the kernel contiguous allocator (the magic of videobuf2-dma-contig.h), and then get VCSM to wrap that into a reloc handle.
Fencing is going to be a nightmare, I suspect. Generally the approach taken was that buffer ownership went with buffer passing. It may be possible to make the kernel side of the V4L2 deal with fences as the buffers come back from the VPU, but I don't know for sure. First I'd have to investigate how fences worked with V4L2 at all!

A way down the line, but how were you anticipating getting the 3D side to consume YUV data, as AFAIK it will only accept RGB? It was one of the very troublesome areas with the GLES driver, and that was on on the VPU. Deal with the HVS first though, as that will quite happily swallow it.

One of Phil's team from back in the day wrote the sysman and clockman remote drivers, so if I can get it working then he and Dom can probably be persuaded. Making it backward compatible so that a new firmware with old kernel (missing the appropriate receiver) works may prove trickier.
Do you have access to the MPS Android kernel tree? Pi Towers have the VPU side of the driver, but not the kernel side. Getting hold of that may save some time if we want to go down this route.

Thanks for the feedback. My playtime is limited by young family, but knowing what is useful helps me prioritise. I will apologise in advance as I'm likely to overcommit myself for what I can actually get done.

anholt, 2016-08-02:
The GL driver already has to do a bunch of shader customizations based on texture format, so inserting sample-3-times-and-colorspace-convert isn't going to be that bad.  The math for the specific tiled format might be a pain, though.

I haven't ever seen that Android tree and don't know where I'd find it.

anholt, 2016-08-17:
@6by9 Have you had any chance to work on the V4L2 stuff? I'm trying to work on the vc4 side of things using some *terrible* hacks to vchiq so the camera can be used with vb2-dc, but zero-copy should be better for everybody.

anholt, 2016-08-17:
So, it appears that vc4 is ready to take dmabuf imports, even without new synchronization work.   With [crazy hacks to the camera module](https://github.com/anholt/linux/tree/rpi-4.4.y-camera-dmabuf) and some [demo code](https://github.com/anholt/drm-v4l2-test), with the command line:

    ./dmabuf-sharing -M vc4 -i /dev/video0 -f BGR4 -F XB24 -b 2

I've got fullscreen camera output at the console.

anholt, 2017-01-27:
Heads up: I just submitted the camera driver as of a15ba877dab4e61ea3fc7b006e2a73828b083c52 upstream to the staging tree.  This was a request from the staging maintainer so that we have the consumer of VCHI in-tree whlie people are actively fixing up VCHI.

gohai, 2017-01-29:
@anholt Exciting news!

I stumbled upon an unrelated issue with the v4l2 camera driver and GStreamer 1.10, where the sizeimage provided by TRY/S_FMT does not match what GStreamer was expecting. [Here](https://bugzilla.gnome.org/show_bug.cgi?id=775564) are the details, including a workaround for GStreamer that does the trick for me. Hoping that this will get merged into future versions of GStreamer, but would it maybe make sense to add to the TODO for staging? (I have no experience with v4l2 myself, unfortunately.)

6by9, 2017-01-29:
This is not really the place to discuss unrelated issues - raise a bug on raspberrypi/linux, but a quick explanation.
It's not a bug. The GPU has a restriction that the height must be a multiple of 16. Your height of 200 is not. The buffer size has to reflect the padded height of a frame to 640x208, otherwise you'll corrupt the memory after your image. 640*208 = 133120 which is the result you have seen.
I've not seen a mechanism within V4L2 to allow a driver to specify vertical padding requirements except by requesting a bigger buffer than is required. This doesn't work on formats such as V4L2_PIX_FMT_YUV420 where you want to specify padding between the planes, so the GPU has to do some further munging of the data on those formats.

gohai, 2017-01-29:
@6by9 Thank you for your explanation and sorry for bringing this up here. I will pass on your info to the GStreamer developers, and I hope they'll handle this in future 1.10.x releases.

gohai, 2018-05-01:
@6by9 Do you have an estimate when the updated v4l2 camera driver will land in Raspbian? Any way to test this currently? Thanks!

6by9, 2018-05-02:
The https://github.com/6by9/linux/tree/rpi-4.14.y-v4l2-codec being developed for #13 has all the required mods for the camera too. It'll all get merged together, and is also hit by the same vcsm-cma cleanup issue.

Testing - it's no real difference from before. Eric's [demo code](https://github.com/anholt/drm-v4l2-test)  he linked to [above](https://github.com/anholt/linux/issues/12#issuecomment-240482532) should still be valid for almost all pixel formats (not YUYV and variants as the hardware doesn't support them).
I haven't got a Pi with camera module set up at the moment, but I'd expect GStreamer's ```v4l2src io-mode=dmabuf``` to just work with it when sending it to kmssink.