anholt/linux #13: vc4: dmabuf import from video decode driver not supported

anholt, 2016-02-15:
We need this for zero-copy display of video.

PTS93, 2016-02-21:
+1
(+1 one anything regarding video decoding actually)

edgmnt, 2017-03-21:
Any updates on this?

Actually, can we get video decoding without zero-copy? I'm thinking perhaps there should be a separate issue to track video decoding per se, here or in /mesa, e.g. VDPAU support.

gohai, 2018-05-01:
@6by9 Do you have an estimate when the updated v4l2 decode driver will land in Raspbian? Any way to test this currently? Thanks!

6by9, 2018-05-02:
https://github.com/6by9/drm_mmal demos using MMAL to decode and libdrm to display (so command-line only, not in X).
@anholt took that and added X support in https://github.com/anholt/drm_mmal/tree/x11
So video decode can be demoed as it stands.

The second step is that more things have V4L2 M2M support for codecs (eg FFmpeg and Gstreamer).
https://github.com/6by9/linux/tree/rpi-4.14.y-v4l2-codec is my working branch for wrapping the MMAL decoder (and encoder) as a V4L2 device.
This is working except for:
- end of stream handling (I've got a slight bodge that has a race condition, but will be looking at fixing it properly)
- there is a cleanup issue in vcsm-cma sometimes if the app is terminated ctrl-c. A refcount somewhere is either not being taken, or released twice, leading to a grumpy kernel.

The LibreElec guys are working on patches for FFmpeg to support dmabufs out.
GStreamer 1.13.90 just works (I haven't tried earlier versions, nor synced since March) and running it into kmssink with dmabufs seems fine.
There is ongoing work to see if we can convince Chromium to use it too.

Timescales for merging the driver to the Pi kernel - hopefully only a few weeks, but it depends on how tricky the cleanup issue is to solve. The Raspbian kernel then gets bumped at a point where it is viewed to be worthwhile and has been proved to be stable.
The more "entertaining" issue is that this all really ought to be upstreamed to the staging tree, and that could be painful.

6by9, 2018-05-02:
EOS support sorted, so just the refcounting in vcsm-cma to resolve.

LibreElec ffmpeg patches at https://github.com/ldts/FFmpeg, with simple demo app at https://github.com/BayLibre/ffmpeg-drm (yes it is meant to flicker as they aren't keeping track of a current buffer, so are adding and then immediately removing the buffer).

stschake, 2018-05-02:
@gohai that's possible through GL_TEXTURE_EXTERNAL_OES; see the X11 example:

https://github.com/anholt/drm_mmal/blob/x11/drm_mmal.c#L443

As the brcm driver did a recent Mesa version can also sample from YUV textures.

PTS93, 2018-05-03:
As a pure user who doesn't understand all too much of how the kernel development and the interplay between all these software pieces works. Am I understanding that correctly that with this V4L2 M2M support I can simply play a video via gstreamer without any plugins and it will be decoded via the VPU?
The gst-omx plugin had the downside that it wasn't fast enough for 1080p playback and you were stuck with 720p in programs like processing that used gstreamer to get video into opengl textures.
Does this also mean the nasty "bug" that you couldn't use USB audio interface for audio output is also a thing of the past with this work?
Is there an ETA when can the general public can try these new drivers out?

6by9, 2018-05-03:
> As a pure user who doesn't understand all too much of how the kernel development and the interplay between all these software pieces works. Am I understanding that correctly that with this V4L2 M2M [...]
The gst-omx plugin had the downside that it wasn't fast enough for 1080p playback and you were stuck with 720p in programs like processing that used gstreamer to get video into opengl textures.

GStreamer was doing funny things with pushing images into GL textures, and that requires significantly more processing than just passing them to the video composition engine (GL requires RGB, video decode produces RGB). It was also slightly inefficient in that it required buffers in ARM memory, when the OMX components require them in GPU memory, therefore there was a bunch of copying going on.

This work is to use dmabufs, which are the Linux kernel mechanism for sharing buffers between subsystems. As long as you select the io-mode on the GStreamer components to be dmabuf then almost all the copying disappears (1080P decode and render via the kmssink component was taking about 2% of the ARMs. I haven't tried via GL).
There may be a requirement to get Raspbian to bump the shipped version of GStreamer in order to work with this, but that is plausible.

> Does this also mean the nasty "bug" that you couldn't use USB audio interface for audio output is also a thing of the past with this work?

I don't know the details of this bug.  If it relates to USB bandwidth issues then this has nothing to do with the USB subsystem.

> Is there an ETA when can the general public can try these new drivers out?

Anyone is free to download my kernel tree, build it, and play right now.
Until at least a few people have tried it out then it won't be merged into the main kernel even though it requires a dtoverlay to load it.

PTS93, 2018-05-03:
Thanks a lot for this insight!
Regarding USB audio, the issue was the omx-api only allows outputting audio via HDMI or the internal PWM audio because it didn't support Alsa. This sounds like it wouldn't suffer from that.