anholt/linux #17: vc4: Do a hilbert walk on the tiles.

anholt, 2016-02-15:
Currently we call our tiles in raster order. If you do a hilbert fractal-like walk instead it can slightly increase cache locality for the texture sampler.

varadgautam, 2016-02-22:
i.e. should `vc4_create_rcl_bo` differentiate between T and LT-formats when writing tiles into the list?

anholt, 2016-02-22:
Nope, you still need to write out your surface the same.  Imagine going from one tile to its neighbor -- you're probably mostly reloading your texture cache because there's different content being sampled in the new tile, except that a few of the pixels near the shared edge might be able to share samples loaded from filtering from the other side of the edge.  Right now we move to the neighbor tile to the right every time, then start at a new scanline of tiles once we reach the end (which probably doesn't share any samples).  If we instead do a hilbert walk (or potentially just swap between left-to-right and right-to-left), we might get that small benefit of cache locality from your neighbor tile a few more times.

varadgautam, 2016-02-23:
I'm sure I'm missing something - the manual mentions that "In tile rendering mode, the change from one tile to the next is handled in a pipelined fashion in the hardware". Is the order of processing tiles the same as the order we place them in the rcl for that frame (= order of calling `emit_tile(x,y)` for a job)?

If not, what determines the order?