anholt/linux #7: vc4: HDMI display fails at resolution other than 1920x1080

anholt, 2016-02-15:
Previous debugging session indicated that the clock driver was failing at <100Mhz, and most non-1920x1080 resolutions were below that.  However, 2016-02-15 debugging indicated that the clock driver was doing OK, but *all* non-1920x1080 resolutions were broken.

ric96, 2016-02-15:
Hmmm... Not very sure about the, the android builds with vc4 enabled run at 1280x720. Maybe its just with raspbian.

anholt, 2016-02-16:
Two separate bugs here: changing modes on upstream driver fails entirely due to missing vertical timing setup, and PLL divider writes weren't latching due to missing CM_PASSWORD (limiting the range of working clock frequencies available).

imzyxwvu, 2016-02-17:
I built an Android M image using peyo-hd's manifests. My monitor and TV both reports invalid input after the rainbow screen, but it seems the system is still starting. On Ubuntu with this driver and Xorg modesetting driver, the resolution also has to be 1920x1080. It's not just with raspbian.

Goofybud16, 2016-02-23:
+1

My 1360x768 display fails.

Any VT before X just shows "No Input" (on the screen, typically shown when there is no device connected) and when X starts, it shows "Not Supported."

This is on the latest version of Raspbian with the Raspbian 4.1 Kernel, if that helps at all.

randyoo, 2016-03-09:
My monitor, a "Hanns-G HW191D" gives an "out of sync" error, and fails to set the normal resolution upon rebooting. (the NOOBS restore screen shows at what appears to be 640x480)
The monitor has to unplugged from the wall to recover!

LordShion, 2016-03-29:
Hi everyone, I seem to run in this problem also. I'm writing here , if it can help for analysing problem
when booting without vc4, everything goes fine changing resolution , & overscan features.
but when booting whit vc4, only 1920x1080 seems to display with no overscan features, other resolutions won't display
futhermore, the raspberry boots normaly and i can connect via ssh.
thus permitting me to do some 
tvservice -o 
and 
tvservice -e 'CEA 4' for example
 (1280x720p), then display stills black, but when I do tty switch(ctrl+alt+F1..F7), it finally displays in correct resolution!!
then doing fbset, tells me geometry is 1920x1080  even if my config.txt's framebuffer size is set to 1280x720.
I'm stuck.

anholt, 2016-04-20:
Pull requests for rpi-4.4 and rpi-4.5 should have all the upstream fixes necessary now.

closed 2016-04-20 by anholt