anholt/linux #19: dtoverlay=vc4-kms-v3d with HDMI unplugged causes boot failure

greyltc, 2016-03-11:
My Pi2 B v1.1 fails to boot when I have `dtoverlay=vc4-kms-v3d` in my config.txt unless an HDMI display is plugged in. This is not solved even when I have the official display connected. This bug seems to be present both in the latest released rasbian and in a kernel I built just now from the rpi-4.1.y branch [here](https://github.com/raspberrypi/linux).

schnitzeltony, 2016-03-11:
Isn't this the same as https://github.com/anholt/linux/issues/11?

greyltc, 2016-03-11:
I don't know. That issue title confuses me: "vc4: HDMI power on when not connected at boot not supported" What's "HDMI power on"?

I actually don't care too much about my Pi not booting when the VC4 overlay is turned on and there's no screen attached (what would I be trying to accelerate anyway in that case?). I would very much like the VC4 acceleration when using the official 7in touchscreen display.

Whatever connecting an HDMI display does that allows the kernel to boot  should also be done by the official 7 in display right?

anholt, 2016-03-11:
I don't think there's anything here not covered in #8 and #11 already.

closed 2016-03-11 by anholt

greyltc, 2016-03-12:
Ah, great to see it's covered. Sorry for the dup'