anholt/linux #20: VC4 breakage at bootup

grbd, 2016-03-12:
Hi,
I recently raised an issue on the rpi linux sources repo, but I think this might be the better place for it
https://github.com/raspberrypi/linux/issues/1344

One of the recent changes there, seems to prevent the kernel from booting up when dtoverlay=vc4-kms-v3d is enabled, commit id 8924ecf62e9ee8ec991d6361152e17268cc7ed6f

that being said without that commit my system will bootup but vc4 is having problems recognising the monitor for some reason (I'm still in the process of trying different options from the default to see if I'm missing something that needs to be compiled in directly)
```
[drm:vc4_hdmi_bind] *ERROR* Failed to get pixel clock
[    2.228836] vc4-drm soc:gpu@7e4c0000: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517
[    2.242500] vc4-drm soc:gpu@7e4c0000: master bind failed: -517
```

perhaps this is related to the issue of the system not booting when there's no monitor connected that others have been seeing

anholt, 2016-03-14:
If you don't have the clock driver, then we won't be able to get clocks.  You need the clock driver.

If you have troubles booting the current branch, you're going to need to hook up a serial console to find what's going wrong.  If you're trying to boot without HDMI connected, you'll probably need the HDMI power on series.

closed 2016-03-14 by anholt

grbd, 2016-03-14:
I'm booting with a hdmi cable connected to a monitor, it's just for some reason it stalls during the boot process. The lack of a clock must be allowing it to continue some how (although be it in a broken state).
I think like you've said a uart cable is probably the only way forward