anholt/linux #22: Gpio capture output / Pixel Clock issue in rpi sources

grbd, 2016-03-14:
I managed to attach a serial cable to the gpio of an rpi2 (I've also got an rpi3 handy)
and captured some logs from the serial console, not sure if these might be handy

I've tried 2 different kernels so far
 * Latest rpi sources now based on full 4.5 release / latest commit - rpi-4.5.y
 * The kernel from the drm-vc4-4.5 branch (it was only updated a few hours ago, so I wasn't expecting anything to work)

The strange thing is, I'm still getting the same error as before about the pixel clock, but since it's the latest commit it should be included in the kernel, I have a feeling it's currently not being linked into the kernel binary, so I'm going to try and look at that next.

  * [rpi-sources No Monitor attached](https://raw.githubusercontent.com/ASoftTech/Appst.Gentoo.Rpi/master/Docs/KernelBuild/dmesg-serial-logs/rpi-sources-nomonitor.txt)
  * [rpi-sources Packard Bell Monitor via DVI](https://raw.githubusercontent.com/ASoftTech/Appst.Gentoo.Rpi/master/Docs/KernelBuild/dmesg-serial-logs/rpi-sources-packardbell-dvi.txt)
  * [rpi-sources Samsung Monitor direct HDMI](https://raw.githubusercontent.com/ASoftTech/Appst.Gentoo.Rpi/master/Docs/KernelBuild/dmesg-serial-logs/rpi-sources-samsung-direct.txt)
  * [rpi-sources Samsung Monitor via hdmi switch box](https://raw.githubusercontent.com/ASoftTech/Appst.Gentoo.Rpi/master/Docs/KernelBuild/dmesg-serial-logs/rpi-sources-samsung-swbox.txt)
  * [rpi-sources LG TV hdmi input](https://raw.githubusercontent.com/ASoftTech/Appst.Gentoo.Rpi/master/Docs/KernelBuild/dmesg-serial-logs/rpi-sources-lgtv-hdmi.txt)
  * [drm-vc4-4.5 dmesg log](https://raw.githubusercontent.com/ASoftTech/Appst.Gentoo.Rpi/master/Docs/KernelBuild/dmesg-serial-logs/aholt.txt)

  * [kernel .config](https://raw.githubusercontent.com/ASoftTech/Appst.Gentoo.Rpi/master/Docs/setup/config.txt)
  * [cmdline.txt](https://raw.githubusercontent.com/ASoftTech/Appst.Gentoo.Rpi/master/Docs/setup/cmdline.txt)
  * [config.txt](https://raw.githubusercontent.com/ASoftTech/Appst.Gentoo.Rpi/master/Docs/setup/config.txt)

grbd, 2016-03-15:
I just tried modifying the Makefile to add in clk-bcm2835-aux but it doesn't seem to have made any difference
I just seem to be getting
```
[    1.996534] vc4-drm soc:gpu@7e4c0000: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517
[    2.005051] vc4-drm soc:gpu@7e4c0000: master bind failed: -517
```

```
--- a/drivers/clk/bcm/Makefile
+++ b/drivers/clk/bcm/Makefile
@@ -5,7 +5,7 @@ obj-$(CONFIG_CLK_BCM_KONA)      += clk-bcm281xx.o
 obj-$(CONFIG_CLK_BCM_KONA)     += clk-bcm21664.o
 obj-$(CONFIG_COMMON_CLK_IPROC) += clk-iproc-armpll.o clk-iproc-pll.o clk-iproc-asiu.o
 obj-$(CONFIG_ARCH_BCM2835)$(CONFIG_ARCH_BCM2708)$(CONFIG_ARCH_BCM2709) += clk-bcm2835.o
-obj-$(CONFIG_ARCH_BCM2835)     += clk-bcm2835-aux.o
+obj-$(CONFIG_ARCH_BCM2835)$(CONFIG_ARCH_BCM2708)$(CONFIG_ARCH_BCM2709) += clk-bcm2835-aux.o
 obj-$(CONFIG_COMMON_CLK_IPROC) += clk-ns2.o
 obj-$(CONFIG_ARCH_BCM_CYGNUS)  += clk-cygnus.o
 obj-$(CONFIG_ARCH_BCM_NSP)     += clk-nsp.o
```

anholt, 2016-04-20:
I've sent a pull request for rpi-4.5.y at this point.  If you're using 2709, you need to be using bcm2709_defconfig on that pull request and the vc4-kms-v3d overlay and avoid_warnings=2 in config.txt.

closed 2016-04-20 by anholt