anholt/linux #23: vc4-drm soc:gpu@7e4c0000: failed to allocate buffer with size 1077248

cneira, 2016-03-25:
When trying to play 0ad, the game load fine but I start a game, at the middle of the loading screen the screen gets garbled and goes to black.
The following messages are found in dmesg

80710.527173] vc4-drm soc:gpu@7e4c0000: failed to allocate buffer with size 5595136
[80710.527467] vc4-drm soc:gpu@7e4c0000: failed to allocate buffer with size 5595136
[80710.527607] vc4-drm soc:gpu@7e4c0000: failed to allocate buffer with size 5595136
[80710.527737] vc4-drm soc:gpu@7e4c0000: failed to allocate buffer with size 5595136
[80710.527782] [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
[80710.527796] [drm] num bos allocated: 694
[80710.527809] [drm] size bos allocated: 245204kb
[80710.527821] [drm] num bos used: 692
[80710.527833] [drm] size bos used: 234276kb
[80710.527845] [drm] num bos cached: 2
[80710.527859] [drm] size bos cached: 10928kb

80715.107110] vc4-drm soc:gpu@7e4c0000: failed to allocate buffer with size 1056768
[80715.107256] vc4-drm soc:gpu@7e4c0000: failed to allocate buffer with size 1056768
[80715.107306] [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
[80715.107322] [drm] num bos allocated: 707
[80715.107335] [drm] size bos allocated: 239508kb
[80715.107350] [drm] num bos used: 701
[80715.107362] [drm] size bos used: 236148kb
[80715.107374] [drm] num bos cached: 6
[80715.107385] [drm] size bos cached: 3360kb
[80715.107425] [drm:vc4_validate_bin_cl [vc4]] *ERROR* 0x00000000: packet 112 (VC4_PACKET_TILE_BINNING_MODE_CONFIG) failed to validate
[80715.112874] vc4-drm soc:gpu@7e4c0000: failed to allocate buffer with size 1060864
[80715.113139] vc4-drm soc:gpu@7e4c0000: failed to allocate buffer with size 1060864
[80717.076550] [drm] Resetting GPU.
[80717.077174] vc4-drm soc:gpu@7e4c0000: failed to allocate buffer with size 1060864
[80717.077312] vc4-drm soc:gpu@7e4c0000: failed to allocate buffer with size 1060864
[80717.077357] [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
[80717.077371] [drm] num bos allocated: 703
[80717.077384] [drm] size bos allocated: 237968kb
[80717.077397] [drm] num bos used: 701
[80717.077409] [drm] size bos used: 235896kb
[80717.077420] [drm] num bos cached: 2
[80717.077432] [drm] size bos cached: 2072kb
[80717.077470] [drm:vc4_validate_bin_cl [vc4]] *ERROR* 0x00000000: packet 112 (VC4_PACKET_TILE_BINNING_MODE_CONFIG) failed to validate

cneira, 2016-03-25:
I have attached the dmesg file when the error happened. after that I cannot see my screen (X is garbled)  nor the virtual consoles. Is there any documentation about the VC4 specs ? I want to learn how I could fix it myself

anholt, 2016-09-29:
0ad is allocating way too much memory to be playable on this hardware.  I've been keeping this bug report open as the placeholder for "I should make memory allocation degrade more  gracefully.

cneira, 2016-09-29:
Hi, 
Could you explain how are your plans on how to accomplish this ?. I have no experience on the VC4-drm code, and I will like to learn how this could be resolved, if you have time to explain your approach.
Bests

anholt, 2016-09-29:
When CMA alloc fails, go through our other BOs, and move them out of CMA and into a shmem file to free CMA space until CMA alloc succeeds.  It will be brutally slow.

cneira, 2016-09-29:
Thank you very much Eric.

closed 2017-03-02 by anholt