anholt/linux #24: Rpi3 won't work on specific displays

ric96, 2016-04-04:
It seems that when i enable open gl on my pi3 it gives me a blank display every time, after some googling i found this forum post: https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=140865&p=936901

so it seems that pi3 crashes / blanks display on some monitors, i do not have another monitor to check on so i am relying on what the forum post says.

Also no UART output so i can't interact with the pi at all.

seniordale, 2016-04-06:
+1, I'm having the same issue. The EDID information is either messed up, or something.

Xorg.0.log:
http://pastebin.com/3SbXUdkb

dmesg:
http://pastebin.com/RHALEPYj

Thanks everyone, looking forward to goofing around with the accelerated graphics! Also, I’m going to try this on my other monitor to make sure it's the display.

ric96, 2016-04-09:
Confirmed EDID Issue but their is a work around:

sudo /opt/vc/bin/tvservice -d /boot/edid.dat

and add hdmi_edid_file=1 to config.txt.

pherelad, 2016-04-09:
Sweet, thanks for the update ric96. I'll give that a shot too and see if I can get past the black screen.

ric96, 2016-04-09:
Ok I found something VERY VERY odd...
The EDID fix is useless :(
My pi3 worked by changing the PSU (WHAT!!!)
I used to run it on a 5v 2A Generic power supply and that causes the gl driver issue...
the second psu is a ASUS branded 5v 1.3A and the gl diver works properly.

pherelad, 2016-04-09:
Oh wierd! I switched PSU's as well, but I didn't try going lower. I actually tried throwing more current at it. I'll see what other ones I have laying around. Thanks for the FYI @ric96 :smile:

pherelad, 2016-04-09:
Since your issue is resolved, perhaps I'll just move my stuff into it's own issue. By the way, could you post the errors (if any) that you were getting in your Xorg.0.log, and in dmesg before you switched PSU's?

anholt, 2016-04-20:
If just changing the power supply fixes it, then it won't be a driver issue.  A weak power supply would be especially a problem if you're missing avoid_warnings=2, which raspi-config fails to set.

Changing the firmware's EDID options also shouldn't affect the driver, since it doesn't use the firmware.

closed 2016-04-20 by anholt

Taxicletter, 2016-04-23:
My Raspberry Pi works perfectly on my Dell display. But after some update, the OpenGL driver doesn't work anymore. I can't increase the GPU memory anymore, it switches back to 64MB each time.
Could the fix you propose also help for that problem?

ric96, 2016-04-23:
yes the gl driver wont work on 4.4.y kernel update, what you can do is
```
sudo BRANCH=next rpi-update
```

Taxicletter, 2016-04-24:
Yes! That worked!

For the future: is it better to use `sudo BRANCH=next rpi-update`?
Or I keep doing `sudo rpi-update`

**Thanks a lot!!**

ltrifonov, 2016-06-07:
The discussion is quite old, but let me share my experience with OpenGL and black screen:

After enabling OpenGL, the rainbow/blackscreen happens, BUT:
 - the PI is actually working, it is accessible via ssh;
 - even though the screen is blank, you can still blindly type `startx` and the X will start and visible;
 - black screen is happening on text modes. Graphical works with some strange not matching resolution;

Hope this will help someone

kenb-sc, 2016-09-09:
ltrifonov mentioned a "strange not matching resolution".  
By measuring and comparing the dimensions of a terminal window with and without OpenGL enabled, I found that the effective screen resolution with OpenGL became
   1150 x 866 (4 x 3 ratio)
versus the native resolution of 
   1920 x 1080 (16 x 9 ratio)
This was with a Raspberry Pi 3 and a Dell S2330MX monitor.

josephdfarkas, 2017-02-08:
Thank you ric96, I spent 2 days looking for the solution. You're a lifesaver!