anholt/linux #25: Raspberry PI 2 yields black screen with EDID errors in logs

pherelad, 2016-04-09:
After enabling the new opengl driver on my Raspberry PI 2, I receive a black screen on my TV after the rainbow. When I SSH in see what's in the logs, I see a bunch of errors about EDID information being goofed up. I've included these logs below for reference.

Xorg.0.log:
http://pastebin.com/3SbXUdkb

dmesg:
http://pastebin.com/RHALEPYj

config.txt:
http://pastebin.com/Pz8bUj7c

Thanks everyone, looking forward to goofing around with the accelerated graphics! Also, I’m going to try this on my other monitor to make sure it's the display.

* I've verified I'm using a 5v 2A power adapter and am running a Raspberry PI 2 with a fresh Raspian image.
* I've tried forcing 1080p using the config.txt file
* I've tried changing the GPU memory
* I've tried ignoring EDID in the config.txt file
* I've tried enabling HDMI boost (4)

anholt, 2016-04-20:
config.txt settings affect the closed-source firmware, not the open driver.  The only extra change you need to make to the firmware is avoid_warnings=2 in config.txt, to keep the firmware from breaking open graphics.

I can't do much with your logs because you trimmed dmesg, so I don't know what kernel you're using.  Also, just to be sure: When you say "TV" you mean "A TV connected through HDMI", right?

pherelad, 2016-04-21:
Hi @anholt, thanks for looking into this.

> config.txt settings affect the closed-source firmware, not the open driver. The only extra change you need to make to the firmware is avoid_warnings=2 in config.txt, to keep the firmware from breaking open graphics.

That's good to know. I'll drop the avoid_warnings=2 into config.txt and see what happens.

> I can't do much with your logs because you trimmed dmesg, so I don't know what kernel you're using. Also, just to be sure: When you say "TV" you mean "A TV connected through HDMI", right?

Ah, sorry about that. I'll throw the full dmesg log up tonight so you have more to go on. As far as the "TV" comment, yes, it's a 42" TV connected via HDMI.

ryancito02, 2016-05-15:
Any progress on this? It seems extremely similar to the problem I am having.

https://github.com/anholt/linux/issues/28

Did you fix it by adding avoid_warnings?

anholt, 2016-08-10:
@pherelad Please attach the full dmesg and Xorg log, not snippets.

WallyCZ, 2016-09-08:
@anholt It seems, that I have same problem:

dmesg:
http://pastebin.com/r5Vzdf5E

Xorg.0.log:
http://pastebin.com/NNA8tUU3

anholt, 2016-09-29:
@pherelad seems to have dropped the bug report, so I'm closing it.  @WallyCZ, your logs don't look anything like the ones from the submitter.  Please open your own bug report (those EDID failures look interesting) with full information about your hardware setup.

closed 2016-09-29 by anholt