anholt/linux #26: VC4 + udlfb

paines, 2016-04-12:
Sorry for my ignorance, but it seems that the VC4 drivers doesn't work quite properly together with pluggable USB display devices, or maybe this is an issue with Xorg. 
I used a Raspberry PI2 + Lenovo LT1412 , but  I guess it should do any other USB display. Without the option "experimental GL driver for desktop" it works quite well, but no OpenGL output on X.

anholt, 2016-04-20:
I don't have any UDL devices, so I don't really have a way to test this.  It may end up getting fixed as part of camera/video decode support, thought.

anholt, 2016-10-28:
You'll probably want to look into something like https://wiki.archlinux.org/index.php/DisplayLink for how to set up UDL these days.  Ultimately, this is not a vc4 driver issue, just Xorg configuration for UDL, so closing.

closed 2016-10-28 by anholt