anholt/linux #27: Screen flickers/works intermittently at default 1920x1080 resolution

joaodriessen, 2016-04-25:
Hi Eric,

Amazed at the progress you're making.
I have an issue that the screen is black at the default 1920x1080 resolution. infrequently i get a picture for about a second and then it disappears again, setting the raspberry pi to output a resolution of 1280x720 does work.

by the way, probably a separate bug (and can file separately if you like) changing the screen resolution via /opt/vc/bin/tvservice causes a purple line(s) to appear at the screen edge(s).

I would like to post any relevant info you need to catch what's going on but I don't know what it is you need.

I'm running on a raspberry pi 3 with arch:
Linux alarmpi 4.4.8-1-ARCH #1 SMP Fri Apr 22 18:47:51 MDT 2016 armv7l GNU/Linux

config.txt:
#hdmi_group=1
#hdmi_mode=4
#disable_overscan=1
#framebuffer_depth=32
#gpu_mem=256
#dtoverlay=hifiberry-dacplus
dtoverlay=vc4-kms-v3d
avoid_warnings=2

The monitor i'm using works fine with other devices at 1920x1080 60/59mhz.

So grateful for your work!

anholt, 2016-05-04:
If you ask the firmware to change graphics settings at runtime, that will completely trash the open driver.  You need to use the normal xrandr command, or whatever your desktop environment has as a friendly GUI, instead.

First step is attaching the dmesg, to make sure the clk driver isn't complaining about anything.  We would probably want to compare /debug/clk/clk_summary to one on a working system at that resolution.

joaodriessen, 2016-05-11:
here's the output of dmesg:

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.9-1-ARCH (builduser@leming) (gcc version 5.3.0 (GCC) ) #1 SMP Fri May 6 12:52:32 MDT 2016
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 256 MiB at 0x10000000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 241664
[    0.000000] free_area_init_node: node 0, pgdat 80bf1200, node_mem_map ba72a000
[    0.000000]   Normal zone: 2124 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 241664 pages, LIFO batch:31
[    0.000000] [bcm2709_smp_init_cpus] enter (9460->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 13 pages/cpu @ba6e6000 s22604 r8192 d22452 u53248
[    0.000000] pcpu-alloc: s22604 r8192 d22452 u53248 alloc=13*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 239540
[    0.000000] Kernel command line: 8250.nr_uarts=1 cma=256M@256M dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2709.boardrev=0xa02082 bcm2709.serial=0x7391aeb4 smsc95xx.macaddr=B8:27:EB:91:AE:B4 bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  root=/dev/mmcblk0p2 rw rootwait console=ttyS0,115200 console=tty1 selinux=0 plymouth.enable=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 kgdboc=ttyS0,115200 elevator=noop quiet loglevel=3 vga=current
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 681320K/966656K available (8303K kernel code, 583K rwdata, 2924K rodata, 440K init, 797K bss, 23192K reserved, 262144K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
                   lowmem  : 0x80000000 - 0xbb000000   ( 944 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x80008000 - 0x80aff22c   (11229 kB)
                     .init : 0x80b00000 - 0x80b6e000   ( 440 kB)
                     .data : 0x80b6e000 - 0x80bfff98   ( 584 kB)
                      .bss : 0x80c02000 - 0x80cc95d0   ( 798 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000020] Switching to timer-based delay loop, resolution 52ns
[    0.000245] Console: colour dummy device 80x30
[    0.000411] console [tty1] enabled
[    0.000446] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000463] pid_max: default: 32768 minimum: 301
[    0.000729] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000742] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001570] Disabling cpuset control group subsystem
[    0.001596] Initializing cgroup subsys io
[    0.001622] Initializing cgroup subsys memory
[    0.001654] Initializing cgroup subsys devices
[    0.001669] Initializing cgroup subsys freezer
[    0.001686] Initializing cgroup subsys net_cls
[    0.001700] Initializing cgroup subsys perf_event
[    0.001715] Initializing cgroup subsys net_prio
[    0.001731] Initializing cgroup subsys pids
[    0.001777] CPU: Testing write buffer coherency: ok
[    0.002223] CPU0: update cpu_capacity 1024
[    0.002234] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002243] [bcm2709_smp_prepare_cpus] enter
[    0.002351] Setting up static identity map for 0x8240 - 0x8274
[    0.003850] [bcm2709_boot_secondary] cpu:1 started (0) 17
[    0.004048] [bcm2709_secondary_init] enter cpu:1
[    0.004086] CPU1: update cpu_capacity 1024
[    0.004092] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[... 421 lines cut ...]
[   13.124321] [<804f70a4>] (dump_stack) from [<8002e7b8>] (warn_slowpath_common+0x84/0xb4)
[   13.124331] [<8002e7b8>] (warn_slowpath_common) from [<8002e884>] (warn_slowpath_null+0x1c/0x24)
[   13.124453] [<8002e884>] (warn_slowpath_null) from [<7f0d92e8>] (drm_update_vblank_count+0x164/0x318 [drm])
[   13.124680] [<7f0d92e8>] (drm_update_vblank_count [drm]) from [<7f0d9500>] (vblank_disable_and_save+0x64/0x74 [drm])
[   13.124897] [<7f0d9500>] (vblank_disable_and_save [drm]) from [<7f0d9574>] (vblank_disable_fn+0x64/0x90 [drm])
[... cut ...]
[   13.125097] ---[ end trace 172ddbe03d05adfc ]---
[   18.829263] fuse init (API version 7.23)
[   19.002880] Bluetooth: RFCOMM TTY layer initialized
[   19.002906] Bluetooth: RFCOMM socket layer initialized
[   19.002926] Bluetooth: RFCOMM ver 1.11
[   29.304873] EXT4-fs (sda3): mounted filesystem with ordered data mode. Opts: (null)
[   68.314578] cfg80211: Verifying active interfaces after reg change

and here's my clk_summary:

[clk_summary.txt](https://github.com/anholt/linux/files/258779/clk_summary.txt)

anholt, 2016-05-11:
>    pllc                                  1            1  2390399926          0 0  
>       pllc_per                           1            1  1195199963          0 0  
>          hsm                             1            1   163620957          0 0  

I think this is it.  The HDMI's been sourced from PLLC, which gets reclocked by the firmware.  I think with https://lkml.org/lkml/2016/4/26/852 (which I should be sending off to rpi-4.4.y soon) you'll get stable video.

joaodriessen, 2016-05-13:
Just noticed the pull request, fantastic!

closed 2016-05-13 by joaodriessen