anholt/linux #28: Black Screen Issue On Boot

ryancito02, 2016-05-02:
I have looked all over for the answer to this, as it seems others have encountered this problem as well, but I could never find a clear answer. Putting "avoid_warnings=2" in my /boot/config.txt did not do anything at all, and using another screen did not work either. I did switch out PSU from a 5V 2.1A to a 5V 2.3A but I still received the red-blue-green-yellow square on bootup, which indicates a brownout, or not enough voltage being supplied. When I had a Raspberry Pi 1, this sign would appear quite frequently and it still worked perfectly, so I don't believe that voltage has anything to do with it.  Anyway, on to the problem. I have a Sandisk 8GB SD Card, which is compatible with the Raspberry Pi 2, according to the website. I flashed the latest Raspbian version (March 2016) onto the SD card using the command "sudo dd bs=4M if=~/Downloads/March2016/raspbian-jessie-March2016.img of=/dev/sdb" and successfully flash Raspbian. I then remove the SD card from the computer, take it out of the adapter, and insert it into my Raspberry Pi 2. It boots up fine, and I reboot it once out-of-the-box to make sure there is no corruption or problems, and after a reboot, I realize everything is working fine. The first things I do is go to my terminal and issue "sudo apt-get update && sudo apt-get -y upgrade" and patiently wait for that to finish. Afterwards, I go to the Raspbian Preferences and click Expand Filesystem, and I reboot. Upon entering Desktop again, I issue "sudo apt-get -y install xcompmgr libgl1-mesa-dri && sudo apt-get -y install libalut0 libalut-dev"
and receive the latest drivers. I issued "sudo raspi-config" and enabled the new OpenGL driver located in Advanced Options. I then reboot my Raspberry Pi 2 in high hopes, and my hopes are dashed by after reboot, a  low voltage / brownout symbol covering the whole screen, after that, a flash of NO SIGNAL and then suddenly, just blackness. A pure black screen. 

I have tried this on two different screens, one 800x600 and another a large screen television to which I don't exactly know the resoulution, but I would guess between 1080p and 1700p. 

Any fix to this?

( I have not tried SSH yet to retrieve the logs, but if you absolutely require the logs, I will do good to enable and get them to you. However, it seems this problem has been answered before, so, I don't really expect to need to give you the logs. If you need them, just ask. )

anholt, 2016-05-03:
Only HDMI is supported: See #8 for tracking DSI panel support.

Yes, logs ar required if we're going to do anything here.  Full dmesg, with drm.debug=0xf on the kernel's command line.  Also probably /var/log/Xorg.0.log.

ryancito02, 2016-05-03:
I am currently using the screen through HDMI, although it supports VGA and AV as well.

As for my logs, here they are:

/boot/config.txt

http://pastebin.com/QZFH2TCu

(The driver is commented out for me to be able to use the Raspberry Pi 2)

/var/log/Xorg.0.log

http://pastebin.com/sB2QbHvU

And here is my Xorg.0.log.old:

http://pastebin.com/ngkaNbNF

and my dmesg output will be in my next comment, since I need to restart
my Raspberry Pi 2 to pass the drm.debug=0xf parameter to my kernel, in /proc/cmdline

Sent from my Raspberry Pi 2, GL driver disabled.

anholt, 2016-05-03:
Your xorg log indicates that the vc4 driver didn't actually load, so the details will be in dmesg.  config.txt is what controls your kernel command line, if you were trying something else.  The dmesg might even be useful without the parameter, though, if it's a probing problem.

ryancito02, 2016-05-04:
I have extracted the log, but I can't seem to copy and paste it through SSH using nano.
I can only get so many lines, here is what I managed to copy and paste.

http://pastebin.com/4v6yzusZ

ryancito02, 2016-05-04:
I have successfully extracted the log from the Raspberry Pi 2 by using the command "dmesg --follow > dmesg.log" and doing a "scp pi@192.168.1.93:~/dmesg.log ~/Desktop" to transfer the file over to my Desktop. Here is the complete log file from the dmesg output:

http://pastebin.com/zEapN4fr

anholt, 2016-05-04:
The dmesg is missing the part from when it was probing modes, and just says that it didn't find any.  You probably want to disable booting into X and just go to console, to shorten the dmesg from starting boot.

ryancito02, 2016-05-04:
Alright, I did an SSH into my Raspberry Pi 2, and did "sudo raspi-config" and made it auto login user pi to console. Strangely enough, when I rebooted, the Raspberry Pi 2 still does not connect to the screen. I assume it's a problem with loading the driver, but this is your forte.

Here is the log from the Console Pi:
http://pastebin.com/fD36NSmL

anholt, 2016-05-05:
Could you do "cat /sys/class/drm/card0-HDMI-A-1/edid > ~/edid" and attach the file?  (there's a link for attaching files at the bottom of the comment box here.  it's really good to use that instead, since pastebins expire)

anholt, 2016-05-05:
Also relevant: if you could plug that monitor into another Linux system with HDMI output and radeon, intel, or nouveau graphics, and get the edid from the corresponding file in their /sys/class/drm/card0, that might narrow things down.

sharpener13, 2016-05-12:
Hi, just feeling this might be related (if not, ignore this message). Having RPI3, Arch Linux 4.4.9-1. When enabling "dtoverlay=vc4-kms-v3d" in config.txt only blank screen is also shown after reboot (then the monitor enters sleep mode, so probably even no signal is generated at all). The (coloured) dmesg outputs attached. It seems there is a stack trace, so something probably went wrong.
[dmesg.vc4.txt](https://github.com/anholt/linux/files/261878/dmesg.vc4.txt)
[dmesg.working.txt](https://github.com/anholt/linux/files/261881/dmesg.working.txt)

ryancito02, 2016-05-15:
The only reason I use Raspbian is because I thought that was the only operating system that the graphics driver was built for. Arch built for Raspberry Pi 2? Please! I have Arch on all my other machines, and to put it on my Raspberry Pi 2 would be great. (I knew there were Arch versions before, but I never knew the OpenGL drivers worked) and now you're saying that the driver would work on Arch? It seems like a Win-Win that I should switch to Arch.

Please, Mr. Sharpener, direct me to the download link for the driver for Arch.

Going to have a blast with my Raspberry Pi 2 now.

ryancito02, 2016-05-15:
After replicating the problem, but with 128 GPU mem, and avoid_warnings=2, I got the huge brownout sign covering the screen, again, and after NO SIGNAL, than a flash of blackness, then "NOT SUPPORT" which I assume means NOT SUPPORTED.

Anholt, does this remind you of any past issues?

anholt, 2016-05-17:
Note: The gpu mem setting in config.txt reserves memory that doesn't get used by the open graphics driver. 

If you got the rainbow screen with the warnings setting in config.txt, then I don't know what's going on.  That should only happen if something used dispmanx, and that's either the undervoltage warning or some ARM program you ran that used dispmanx.  Very recent firmware should prevent anything on ARM from using dispmanx.

ryancito02, 2016-05-17:
It must be an exterior problem..

My screen is 1020 x 600. Is that a problem?

My PSU is 5.0V and 2.1A wall charger, connected to a USB to MICROUSB to my RPI2. 
Is that a problem?

If both of those are not problems, then please, make a tutorial on setting this up correctly, or direct me to one. Thanks!

ryancito02, 2016-05-21:
Solved it, you need not worry.
I looked at Mr. Shapener's comment, and discovered I could run Arch.
So, I booted up Arch according to the Arch Linux ARM website.
I got into Arch, installed myself an LXDE  desktop environment coupled with a login manager, then I went and did a pacman -Syu and installed the correct packages.
I installed raspberrypi-linux, the official raspberry pi linux package, and then I had a /boot/config.txt and the rest of it. I then edited my cmdline.txt to contain cma=256M and my config.txt to contain avoid_warnings=2 gpu_mem=256 and the line of code to enable the vc4 driver. Upon reboot, my login manager popped up, and fitted itself to all the sides of my screen. I logged in and it worked perfectly with GLXGEARS and other 3D OPENGL programs. Thanks!

closed 2016-05-21 by ryancito02