anholt/linux #29: RPi B+ Stuck At Rainbow, No UART Output

173210, 2016-05-20:
Followed https://dri.freedesktop.org/wiki/VC4/, and used `vc4_kms_v3d` branch (5e0242faceacf9fd72f06750e89236ef04e2c372), but I had no luck.
I wrote `cmdline.txt` as the following but it has no UART output.

```
cma=256M@512M console=ttyAMA0,115200 kgdboc=ttyAMA0,115200
```

`config.txt` only contains `avoid_warnings=2\n`.
The firmware is up-to-date (e0fedf15518eef8a56f41113abd5ec8171e3981c), and has no additional bootloader such as Das U-Boot.

I also confirmed rpi-4.4.y(5bc0a41ea6cbee3e720b401350dbdf1b2f08debd) works with the above configuration, but doesn't have DRM.
Any idea?

anholt, 2016-05-20:
The intent wasn't for you to use the vc4_kms_v3d branch.  I've rewritten the instructions to hopefully be clearer.

173210, 2016-05-26:
Thanks. I'll close this issue.

closed 2016-05-26 by 173210