anholt/linux #30: RPi B+ No HDMI Output

173210, 2016-05-21:
I'm using 5bc0a41ea6cbee3e720b401350dbdf1b2f08debd with a DVI monitor (K212HQL)
Here is dmesg.
https://gist.github.com/173210/d6e287bd3c45307a37c081218e6a6c09

```
[    0.879274] [drm] Initialized drm 1.1.0 20060810
[    0.886191] [drm:vc4_hdmi_bind] *ERROR* Failed to get pixel clock
[    0.892419] vc4-drm soc:gpu: failed to bind 20902000.hdmi (ops vc4_hdmi_ops): -517
[    0.900297] vc4-drm soc:gpu: master bind failed: -517
(snip)
[    1.549134] bcm2708_i2c 20805000.i2c: BSC2 Controller at 0x20805000 (irq 77) (baudrate 100000)
(snip)
[    1.765351] vc4-drm soc:gpu: bound 20902000.hdmi (ops vc4_hdmi_ops)
[    1.771939] vc4-drm soc:gpu: bound 20206000.pixelvalve (ops vc4_crtc_ops)
(snip)
[    1.794617] vc4-drm soc:gpu: bound 20207000.pixelvalve (ops vc4_crtc_ops)
[    1.801694] vc4-drm soc:gpu: bound 20807000.pixelvalve (ops vc4_crtc_ops)
[    1.823146] vc4-drm soc:gpu: bound 20400000.hvs (ops vc4_hvs_ops)
[    1.843225] vc4-drm soc:gpu: bound 20c00000.v3d (ops vc4_v3d_ops)
[    1.850835] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[    1.873077] [drm] No driver support for vblank timestamp query.
[    1.879278] vc4-drm soc:gpu: No connectors reported connected with modes
[    1.888631] [drm] Cannot find any crtc or sizes - going 1024x768
[    1.925583] Console: switching to colour frame buffer device 128x48
[    1.955738] vc4-drm soc:gpu: fb0:  frame buffer device
```

So it's failing to detect the monitor?

173210, 2016-05-21:
Here is `vcdbg log msg`.

```
001039.767: HDMI:EDID version 1.3, 0 extensions, screen size 46x26 cm
001039.831: HDMI:EDID features - videodef 0x80 !standby !suspend active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspace; preferred format is nativF
001039.893: HDMI:EDID found preferred CEA detail timing format: 1920x1080p @ 60 Hz (16)
001039.932: HDMI:EDID found DMT format: code 4, 640x480p @ 60 Hz in established timing I/II
001039.972: HDMI:EDID found DMT format: code 8, 800x600p @ 56 Hz in established timing I/II
001040.009: HDMI:EDID found DMT format: code 9, 800x600p @ 60 Hz in established timing I/II
001040.049: HDMI:EDID found DMT format: code 16, 1024x768p @ 60 Hz in established timing I/II
001040.086: HDMI:EDID found DMT format: code 17, 1024x768p @ 70 Hz in established timing I/II
001040.179: HDMI:EDID standard timings block x 8: 0xD1C0 B300 9500 8180 8100 81C0 714F 0101 
001040.243: HDMI:EDID found DMT format: code 82, 1920x1080p @ 60 Hz (16:9) in standard timing 0
001040.295: HDMI:EDID found DMT format: code 58, 1680x1050p @ 60 Hz (16:10) in standard timing 1
001040.346: HDMI:EDID found DMT format: code 47, 1440x900p @ 60 Hz (16:10) in standard timing 2
001040.398: HDMI:EDID found DMT format: code 35, 1280x1024p @ 60 Hz (5:4) in standard timing 3
001040.446: HDMI:EDID found DMT format: code 28, 1280x800p @ 60 Hz (16:10) in standard timing 4
001040.499: HDMI:EDID found DMT format: code 85, 1280x720p @ 60 Hz (16:9) in standard timing 5
001040.548: HDMI:EDID found DMT format: code 21, 1152x864p @ 75 Hz (4:3) in standard timing 6
001040.582: HDMI:EDID moving support for CEA mode 16 to DMT mode 82 because sink does not support HDMI
001040.615: HDMI:EDID filtering formats with pixel clock > 162 MHz or h. blanking > 1023
001040.751: HDMI:EDID preferred mode is updated to DMT (82) 1920x1080p @ 60 Hz with pixel clock 148500000 Hz
001052.117: HDMI:Setting property pixel encoding to Default
001052.135: HDMI:Setting property pixel clock type to PAL
001052.156: HDMI:Setting property content type flag to No data
001052.175: HDMI:Setting property fuzzy format match to enabled
001232.327: hdmi: HDMI:>>>>>>>>>>>>>Rx sensed, reading EDID<<<<<<<<<<<<<
001244.463: hdmi: HDMI:EDID version 1.3, 0 extensions, screen size 46x26 cm
001244.533: hdmi: HDMI:EDID features - videodef 0x80 !standby !suspend active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspace; preferred format isF
001244.597: hdmi: HDMI:EDID found preferred CEA detail timing format: 1920x1080p @ 60 Hz (16)
001244.643: hdmi: HDMI:EDID found DMT format: code 4, 640x480p @ 60 Hz in established timing I/II
001244.683: hdmi: HDMI:EDID found DMT format: code 8, 800x600p @ 56 Hz in established timing I/II
001244.725: hdmi: HDMI:EDID found DMT format: code 9, 800x600p @ 60 Hz in established timing I/II
001244.768: hdmi: HDMI:EDID found DMT format: code 16, 1024x768p @ 60 Hz in established timing I/II
001244.811: hdmi: HDMI:EDID found DMT format: code 17, 1024x768p @ 70 Hz in established timing I/II
001244.904: hdmi: HDMI:EDID standard timings block x 8: 0xD1C0 B300 9500 8180 8100 81C0 714F 0101 
001244.975: hdmi: HDMI:EDID found DMT format: code 82, 1920x1080p @ 60 Hz (16:9) in standard timing 0
001245.031: hdmi: HDMI:EDID found DMT format: code 58, 1680x1050p @ 60 Hz (16:10) in standard timing 1
001245.084: hdmi: HDMI:EDID found DMT format: code 47, 1440x900p @ 60 Hz (16:10) in standard timing 2
001245.138: hdmi: HDMI:EDID found DMT format: code 35, 1280x1024p @ 60 Hz (5:4) in standard timing 3
001245.207: hdmi: HDMI:EDID found DMT format: code 28, 1280x800p @ 60 Hz (16:10) in standard timing 4
001245.264: hdmi: HDMI:EDID found DMT format: code 85, 1280x720p @ 60 Hz (16:9) in standard timing 5
001245.315: hdmi: HDMI:EDID found DMT format: code 21, 1152x864p @ 75 Hz (4:3) in standard timing 6
001245.350: hdmi: HDMI:EDID moving support for CEA mode 16 to DMT mode 82 because sink does not support HDMI
001245.389: hdmi: HDMI:EDID filtering formats with pixel clock > 162 MHz or h. blanking > 1023
001245.530: hdmi: HDMI:EDID preferred mode is updated to DMT (82) 1920x1080p @ 60 Hz with pixel clock 148500000 Hz
001245.549: hdmi: HDMI: hotplug attached with DVI support
001245.616: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
001248.078: hdmi: HDMI: power_on to DMT mode 1080p60
001251.253: hdmi: HDMI: Action callback added to queue to happen at frame 0
001268.200: *** Restart logging
001270.492: Read command line from file 'cmdline.txt'
earlyprintk console=tty0 console=ttyAMA0 root=/dev/mmcblk0p2 rootwait
001542.022: Loading 'kernel.img' to 0x8000 size 0x3e1a74
001545.664: Kernel trailer DTOK property says yes
001545.677: Kernel trailer DDTK property says no
001551.468: Loading 'bcm2835-rpi-b-plus.dtb' to 0x3e9a74 size 0x3143
001595.187: dtparam: pwr_led_gpio=35
001601.218: dtparam: uart0_clkrate=3000000
001636.581: Loaded overlay 'vc4-kms-v3d'
001636.598: dtparam: cma-128=true
001713.462: dtparam: arm_freq=700000000
001715.529: Unknown dtparam 'arm_freq' - ignored
001715.565: dtparam: core_freq=250000000
001722.785: dtparam: cache_line_size=32
001737.980: Device tree loaded to 0x1bfec900 (size 0x36c7)
001739.236: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
003550.232: vchiq_core: vchiq_init_state: slot_zero = 0x48080000, is_master = 1
003554.292: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
003559.378: TV service:host side not connected, dropping notification 0x00000002, 0x00000002, 0x00000052
```

173210, 2016-05-22:
Added `hdmi_ignore_edid=0xa5000080` to `config.txt` and that made some progress.
https://gist.github.com/173210/d6e287bd3c45307a37c081218e6a6c09
```
[    1.849413] [drm:drm_minor_register] 
[    1.850035] [drm:drm_minor_register] new minor registered 64
[    1.850063] [drm:drm_minor_register] 
[    1.850076] [drm:drm_minor_register] 
[    1.850536] [drm:drm_minor_register] new minor registered 0
[    1.850739] [drm:drm_sysfs_connector_add] adding "HDMI-A-1" to sysfs
[    1.850762] [drm:drm_sysfs_hotplug_event] generating hotplug event
[    1.850855] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[    1.863089] [drm] No driver support for vblank timestamp query.
[    1.869203] [drm:drm_helper_probe_single_connector_modes_merge_bits] [CONNECTOR:19:HDMI-A-1]
[    1.869238] [drm:drm_helper_probe_single_connector_modes_merge_bits] [CONNECTOR:19:HDMI-A-1] status updated from 3 to 2
[    1.869258] [drm:drm_helper_probe_single_connector_modes_merge_bits] [CONNECTOR:19:HDMI-A-1] disconnected
[    1.869287] vc4-drm soc:gpu: No connectors reported connected with modes
```

173210, 2016-05-22:
The log had appeared only once. It's failing again. :/

173210, 2016-05-26:
I got it work by following the Wiki, so please close this issue if you `wontfix` or you think it's `invalid`.

anholt, 2016-05-31:
OK, that log just looks like a custom kernel where you missed the power domain driver.  Closing.

closed 2016-05-31 by anholt