anholt/linux #31: typo https://dri.freedesktop.org/wiki/VC4/ causing boot failure?

joaodriessen, 2016-05-24:
grateful for kernel compilation instructions at https://dri.freedesktop.org/wiki/VC4/

after building and installing the kernel I'm having trouble booting it (get's stuck on rainbow screen).
The following may be the issue:

> The kernel will have installed under /boot with some version (make kernelversion tells you what it was). Update your config.txt to contain:
> 
>  avoid_warnings=2     # VPU shouldn't smash our display setup.
>  kernel=vmlinuz-KERNELVERSION
>  device_tree=dtbs/KERNELVERSION/bcm2836-rpi-2-b.dtb

is vmlinuz (second line of config.txt) correct? should it be vmlinux?

and then comes the pasting of correct values for KERNELVERSION

output of make kernerversion is: 4.6.0

but should I replace KERNELVERSION with 4.6.0-next-20160524 (since that's the version I did git checkout for)?

another thing, the dtbs folder doesn't exist in my /boot.

current contents of /boot:
System.map-4.6.0-next-20160524	fixup_db.dat
bcm2709-rpi-2-b.dtb		fixup_x.dat
bcm2710-rpi-3-b.dtb		kernel7.img
bootcode.bin			overlays
cmdline.txt			start.elf
cmdline.txt.backup	start_cd.elf
config.txt			start_db.elf
config.txt.backup		start_x.elf
fixup.dat			vmlinux-4.6.0-next-20160524
fixup_cd.dat

System specs: Raspberry Pi 3 with arch linux ARM installed.

anholt, 2016-05-31:
Fixed up the instructions some more:

It was a typo to say install instead of zinstall.  I also missed
modules_install and dtbs_install.  And "make kernelversion" should have been "make kernelrelease".

Thanks for the feedback!

joaodriessen, 2016-05-31:
Awesome, thanks a bunch!

closed 2016-05-31 by joaodriessen