anholt/linux #32: Render node issue

hjourdain, 2016-05-30:
Hello,

While using the render node, I found that the current implementation in raspberrypi 4.4 kernel lacks some flags allowing the IOCTLs to be used with a render mode.
Contrary to other flags, which are restricters, the flag for render node is an enabler (the IOCTL can't be used from render node if it's not present).
So it seems you need to add DRM_RENDER_ALLOW to all the flags that were previously to 0.
The attached patch is - hopefully - doing that.

Herve
[0002-vc4-ioctl-rendering-allow.patch.txt](https://github.com/anholt/linux/files/289603/0002-vc4-ioctl-rendering-allow.patch.txt)

anholt, 2016-05-31:
If you submitted the patch using git-format-patch (make sure you've got your name and email address) with an appropriate signed-off-by (see Documentation/SubmittingPatches) and drop the "0|" from each, I'd be happy to get it upstreamed with your name on it.

hjourdain, 2016-05-31:
Hello,

 

Please find the new patch attached.

 

Herve

hjourdain, 2016-05-31:
[0001-Add-DRM_RENDER_ALLOW-to-the-flags-for-IOCTLs-so-the-.patch.txt](https://github.com/anholt/linux/files/292045/0001-Add-DRM_RENDER_ALLOW-to-the-flags-for-IOCTLs-so-the-.patch.txt)

anholt, 2016-07-08:
Merged to airlied's drm-fixes:

commit b10c22e5f9902a329450c2027e9291b71e9f1602
Author: Herve Jourdain <herve.jourdain@neuf.fr>
Date:   Wed Jun 1 02:24:46 2016 +0800

closed 2016-07-08 by anholt