anholt/linux #33: Different colors with kernel 4.4.11

anyc, 2016-05-31:
After the upgrade to (a self-built) 4.4.11, I noticed that the colors on my HDMI TV look brighter and light colors become white. I guess this has something to do with the VC4 update and "drm/vc4: Add support for gamma ramps". Hence, I wanted to ask if this is intended? I don't run a DE but just a fullscreen QML-based slideshow with Xorg.

Although xrandr shows me that the brightness is set to 1, setting it again to 1 seems to fix the colors. I noticed though that the gamma also gets changed from 0.46:0.46:0.46 to 1:1:1 although I only set the brightness.

anyc, 2016-05-31:
By the way, I noticed that the slideshow (with moving images) works smoother with the new version, thanks for your work. However, I still get periodic "Resetting GPU" messages in the kernel log and the device freezes after a few hours. Is there a way to check if this is an issue caused by VC4?

anholt, 2016-05-31:
The xrandr --brightness is just applying a transformation to the gamma tables.  If you've played with the tables in some other way (maybe something in your slideshow app?), then xrandr's brightness/gamma values it tries to guess from the table contents will be weird.  If you're wondering what is setting up your tables to something odd, I'd recommend sshing in to the system and using gdb on Xorg with a breakpoint on drmModeCrtcSetGamma before you start your session.

anyc, 2016-08-09:
There was indeed a wrong configuration setting I missed, sorry. My app also works better now with the recent kernel version. Thank you!

closed 2016-08-09 by anyc