anholt/linux #37: Support for EGL (without X11)

loganmc10, 2016-06-08:
Forgive me if I get some of the terminology wrong. I can't seem to get this driver to work without a running X server. Is that how it is supposed to be? Is there support for running applications using EGL without having an X server running?

If so, how would one go about doing things like setting the resolution?

vanfanel, 2016-06-11:
@loganmc10 : EGL without X11 is perfectly possible. SDL2 on the Pi uses dispmanx for the EGL context, and so does RetroArch.
Look in /opt/vc/src/hello_pi for examples.

loganmc10, 2016-06-11:
I was under the impression that the dispmanx stuff doesn't work with this driver, is that wrong?

vanfanel, 2016-06-12:
You are right. I came to post a question myself, saw your question and didn't read the "this driver" part. Sorry.
I am *almost* sure that the MESA driver is compatible with a KMS context, however, so the kmscube should run on the Pi if you boot it in KMS mode instead of dispmanx and have the necessary libs installed.

sirspudd, 2016-06-15:
@loganmc10 The Qt eglfs driver (which has kms provisioning) works great against both the proprietary driver and vc4. I simply switched from one impl to the other in Qt 5.7, and eglfs works perfectly with both driver sets. (It does require a recompile though as the EGL helpers are different)

closed 2016-06-27 by loganmc10