anholt/linux #39: vc4 terminal is lost on Raspberry Pi 3 boot

sirspudd, 2016-06-15:
Hola,

Thank you for the awesome work to date. I followed your instructions to get Linux 4.7 compiled and running on the Pi. This works awesome on the Raspberry Pi 2, but the same sd-card blanks out following the:

"Switching to vc4drmfb from simplefb"

My kernel sources have b3a15f6d, which looks like it should address exactly this issue. (The associated description is perfect.)

Please let me know how I can help resolving this

anholt, 2016-09-06:
This is due to us not having access to the GPIO expander for hotplug detect.  Cherry-picking aa8cbd1ff120c27ce2c3206b1ad57cbf231b4572 from the downstream tree should work around it.

sirspudd, 2016-09-06:
Thank you, I grabbed an earlier one of your patches from the fedora 24 aarch64 kernel sources and hence have been a happy camper for the last couple weeks.

closed 2016-09-06 by sirspudd