anholt/linux #40: vc4-drm: asynchronous atomic pageflipping still not working?

vanfanel, 2016-06-19:
Hi,

Trying to do:
`
drmModeAtomicCommit(drm.fd, req, DRM_MODE_PAGE_FLIP_EVENT|DRM_MODE_PAGE_FLIP_ASYNC, &pipe);`

results on synchronous drmModeAtomicCommit() call: it returns when pageflip is done, instead of returning immediately as it is supposed to do.
So I was wondering if asynchronous atomic pageflipping is supposed to be working already with this KMS/DRM driver or not.
If it's not, I could implement my own thread, but having this block in the main thread is not desired at all for CPU-intensive emulation: we don't have CPU time to waste waiting for vsync.
But I would like to know if it's supposed to be working already.

anholt, 2016-07-03:
Async pageflips should be working, and given the performance boosts I've seen, it seemed like they were.  You probably want to instrument vc4_atomic_commit() to see where you might be blocking.

anholt, 2016-09-06:
My best guess would be that you were running into a lack of commit e7c31f6f25b84fed961dc0dce6248878527693ae in the kernel, and the userspace was submitting the next flip before it should try to.  Closing due to previous lack of feedback, feel free to reopen if you have more debug.

closed 2016-09-06 by anholt