anholt/linux #42: Overscan issue with OpenGL driver.

sgtwilko, 2016-06-30:
Hi
I hope this is the correct place to report this, I have done some searching and not found anything relating to this issue.

When running Raspbian without the OpenGL driver to my Full HD TV (via HDMI) we can see the entire screen, both at bootup and when running X.
When we swap over to the OpenGL driver the image extends past the edge of the TV on all sides, both on the console at bootup and when running X.

Changing the overscan options from raspi-config has no effect.
Changing the overscan_{left|right|top|bottom} and the disable_overscan in config.txt, has no effect.

Am I just missing the obvious, configured wrong, or is this a bug?
Let me know what information you require and I will get it for you.

Thank you for all your hard work getting this driver working :-)

anholt, 2016-07-08:
The open driver scans out pixels 1:1, and it has no way to see the contents of your config.txt

Many TVs default to scaling from a subset of the screen, so rather than telling everyone to go in their TV settings to make it scan pixels 1:1 (or use the "PC" HDMI plug), the config.txt overscan fields make the HVS scale the screen contents down to a smaller area, so that it hopefully lines up with where your TV is scaling up from.  This will hurt image quality, but it's a workaround.  Nouveau and Radeon have underscan KMS properties that do scaling like the old config.txt entries to support those TVs, so I should add those properties too.

sgtwilko, 2016-07-08:
Interesting, I thought I'd configured the TV not to scale at all.
I will double check and get back to you.
Would there be anything else you would need to know?

anholt, 2016-09-30:
Can you attach your xrandr --verbose output?  I'm curious what your CEA block reports.

anholt, 2016-10-10:
With current drm-next, some displays may be better behaved about overscan.

sgtwilko, 2016-10-12:
Hi sorry for the delay, will try to get this to you this week.

sgtwilko, 2016-12-03:
I've attached the results both with and without the opengl driver.

Let me know if you need anything else.

[normalXrandr.txt](https://github.com/anholt/linux/files/628610/normalXrandr.txt)
[openglXrandr.txt](https://github.com/anholt/linux/files/628611/openglXrandr.txt)

deuteragenie, 2019-05-15:
I experience the same issue on Raspbian Stretch 2019-04-08 + Samsung TV.

qwertychouskie, 2019-07-16:
@deuteragenie On some Samsung TVs, labeling the HDMI input used for the Pi as PC (or whatever it is called in their menus) will help fix image issues.

komrad2236, 2021-11-26:
Still not solved. I just installed new Raspberry Pi OS which is Bullseye which has this driver now enabled by default, its not experimental anymore and issue is still here.

If I turn it off, overscan configurations I set in /home/config.txt work just fine but if I enable it, it doesn't work.

I am using RPi 3b+ on CRT tv using composite cable 

I had no issues with Raspbian Buster or any other version of the OS before, but now...

I can't believe it, this is some Microsoft and Bethesda shit right here folks, push non working stuff as default for everyone to use.