anholt/linux #44: (bcm2837-64-next) vc4/drm fail to detect the display

Yamakuzure, 2016-07-08:
Hello from Hamburg!

Thank you very much for your awesome work!

I am currently trying to get the Raspberry PI 3 to boot in 64bit mode with activated VideoCore device, but am still failing.

After some booting using simplefb, the console blanks out.

I have retrieved the dmesg from the failed boot and found these entries:
--------

# grep -i -P "(vc4|drm|console)" dmesg_not_working2.txt
[    0.000000] Kernel command line: console=tty0 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes cma=256M@512M rootwait
[    0.004791] Console: colour dummy device 80x25
[    0.025957] console [tty0] enabled
[    2.380990] Console: switching to colour frame buffer device 210x65
[   28.405139] [drm] Initialized drm 1.1.0 20060810
[   32.569560] vc4-drm soc:gpu: bound 3f902000.hdmi (ops vc4_hdmi_ops)
[   32.684805] vc4-drm soc:gpu: bound 3f206000.pixelvalve (ops vc4_crtc_ops)
[   32.908000] vc4-drm soc:gpu: bound 3f207000.pixelvalve (ops vc4_crtc_ops)
[   33.023165] vc4-drm soc:gpu: bound 3f807000.pixelvalve (ops vc4_crtc_ops)
[   33.131309] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops)
[   33.240326] vc4-drm soc:gpu: bound 3fc00000.v3d (ops vc4_v3d_ops)
[   33.347174] fb: switching to vc4drmfb from simple
[   33.457027] Console: switching to colour dummy device 80x25
[   33.514638] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[   33.515617] [drm] **No driver support for vblank timestamp query.**
[   33.520274] vc4-drm soc:gpu: **No connectors reported connected with modes**
[   33.521854] [drm] **Cannot find any crtc or sizes - going 1024x768**
[   34.070008] Console: switching to colour frame buffer device 128x48
[   34.506082] vc4-drm soc:gpu: fb0:  frame buffer device
--------

There are other issues (like USB not working), which might have to be fixed, first, but maybe you have an idea why the gpu can not find the display?

What information can I provide to help analyzing the problem?

Thank you very much in advance

Sven

anholt, 2016-07-08:
We need a new GPIO driver for the GPIO expander that the HDMI HPD is on.  Sadly, the native driver I wrote doesn't work because the firmware is hitting the I2C to it all the time.

Yamakuzure, 2016-07-12:
Nevertheless you did an awesome job. Unbelievable how far the 64bit stuff is already. Not long ago, even thinking abut activating the VideoCore would make the RPi3 just grin evilly at me. ;-)
I'll just stay tuned and work in 32bit mode, then.

So this is a WONTFIX at the moment, I guess it is in order to close the issue for now.

closed 2016-07-12 by Yamakuzure