anholt/linux #45: (bcm2837-64-next) USB hub does not get activated/detected

Yamakuzure, 2016-07-08:
Hello from Hamburg!

Thank you very much for your awesome work!

I am currently trying to get my Raspberry Pi 3 to boot in 64bit mode using your bcm2837-64-next branch.

While USB works fine using the kernel from https://github.com/Electron752/linux/tree/rpi-4.5.y+rpi364 , it does not work using the most current 4.7 kernel from bcm2837-64-next.

What I get in dmesg is:

--------

 grep -i -P "usb" dmesg_not_working.txt 
[    1.779777] usbcore: registered new interface driver usbfs
[    1.783796] usbcore: registered new interface driver hub
[    1.788265] usbcore: registered new device driver usb
[   29.960789] usbcore: registered new interface driver cdc_ether
[   30.077724] usbcore: registered new interface driver smsc95xx
[   30.337978] 3f980000.usb supply vusb_d not found, using dummy regulator
[   30.452091] 3f980000.usb supply vusb_a not found, using dummy regulator
[   30.696159] dwc2 3f980000.usb: 256 invalid for host_nperio_tx_fifo_size. Check HW configuration.
[   30.809321] dwc2 3f980000.usb: 512 invalid for host_perio_tx_fifo_size. Check HW configuration.
[   31.000247] dwc2 3f980000.usb: Specified GNPTXFDEP=1024 > 32
[   31.116064] dwc2 3f980000.usb: EPs: 8, dedicated fifos, 4080 entries in SPRAM
[   31.253264] dwc2 3f980000.usb: DWC OTG Controller
[   31.366900] dwc2 3f980000.usb: new USB bus registered, assigned bus number 1
[   31.481353] dwc2 3f980000.usb: irq 41, io mem 0x00000000
[   31.633936] hub 1-0:1.0: USB hub found
[   31.909958] usbcore: registered new interface driver usb-storage
[   32.028306] usbcore: registered new interface driver usbled
[   33.181042] usbcore: registered new interface driver usbhid
[   33.288221] usbhid: USB HID core driver

--------

Unfortunately no other USB events show up, so I have neither ethernet, nor keyboard or mouse.

Do you have any idea, what I am missing? I am using the same configuration as with the Electron752 kernel, updated with your current defconfig.

What information can I provide to help analyzing the problem?

Thank you very much

Sven

anholt, 2016-07-08:
You should be using the in-tree defconfig; I have no idea what someone else's config would be doing.

Yamakuzure, 2016-07-12:
Well, I did use your arch/arm64/configs/defconfig. Sorry for not pointing this out right from the start.

lategoodbye, 2016-07-28:
@Yamakuzure In case you still have the problem, try to add dr_mode property in arch/arm64/boot/dts/broadcom/bcm2837-rpi-3-b.dts:

    &usb {
            dr_mode="host";
    };

anholt, 2016-09-29:
Do you still have this problem with linux-next?

anholt, 2016-10-28:
Closing -- lots of people are using upstream arm64 with USB, so presumably this was just due to using a stale development branch.

closed 2016-10-28 by anholt