anholt/linux #46: Pink column of pixels with the KMS/DRM driver

vanfanel, 2016-07-24:
Hi,

Using the old graphics mode, there's no problem, but booting with the VC4 driver shows a pink column of pixels on the console, on the left border of the HDMI TV.
My Pi and TV are running at the recommended mode of 1360x768 pixels.

This is the problem:

[https://www.dropbox.com/s/wdf64hb0b26p8ha/20160724_221438.jpg?dl=0](url)

Any ideas on this issue?

anholt, 2016-08-12:
Can you attach a full dmesg and /sys/class/drm/card0-HDMI-A-1/edid

Due to limitations of EDID, 1366x768 has to be reported as 1360 or 1368 wide, and there are hacks to see this and do the right thing, which might not be doing the right thing on your display.

anholt, 2016-08-12:
(also, if you can start X, the output of xrandr when the bug is present would be interesting)

vanfanel, 2016-08-15:
@anholt:

Full dmesg:

`
pi@raspberrypi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.11-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #888 SMP Mon May 23 20:10:33 BST 2016
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 256 MiB at 0x10000000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 225280
[    0.000000] free_area_init_node: node 0, pgdat 808c0dc0, node_mem_map b67ba000
[    0.000000]   Normal zone: 1980 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 225280 pages, LIFO batch:31
[    0.000000] [bcm2709_smp_init_cpus] enter (9520->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 13 pages/cpu @b6776000 s22592 r8192 d22464 u53248
[    0.000000] pcpu-alloc: s22592 r8192 d22464 u53248 alloc=13*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 223300
[    0.000000] Kernel command line: 8250.nr_uarts=0 cma=256M@256M dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1360 bcm2708_fb.fbheight=768 bcm2709.boardrev=0xa02082 bcm2709.serial=0x80aaa8f0 smsc95xx.macaddr=B8:27:EB:AA:A8:F0 bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 619672K/901120K available (6344K kernel code, 432K rwdata, 1712K rodata, 476K init, 764K bss, 19304K reserved, 262144K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xb7800000 - 0xff800000   (1152 MB)
    lowmem  : 0x80000000 - 0xb7000000   ( 880 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x807e6420   (8058 kB)
      .init : 0x807e7000 - 0x8085e000   ( 476 kB)
      .data : 0x8085e000 - 0x808ca0c8   ( 433 kB)
       .bss : 0x808cd000 - 0x8098c1ac   ( 765 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000009] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000026] Switching to timer-based delay loop, resolution 52ns
[    0.000290] Console: colour dummy device 80x30
[    0.001345] console [tty1] enabled
[    0.001388] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001456] pid_max: default: 32768 minimum: 301
[    0.001794] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001837] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002789] Disabling cpuset control group subsystem
[    0.002847] Initializing cgroup subsys io
[    0.002899] Initializing cgroup subsys memory
[    0.002960] Initializing cgroup subsys devices
[    0.003003] Initializing cgroup subsys freezer
[    0.003046] Initializing cgroup subsys net_cls
[    0.003118] CPU: Testing write buffer coherency: ok
[    0.003208] ftrace: allocating 21205 entries in 63 pages
[    0.052399] CPU0: update cpu_capacity 1024
[    0.052461] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.052496] [bcm2709_smp_prepare_cpus] enter
[    0.052655] Setting up static identity map for 0x8240 - 0x8274
[    0.054323] [bcm2709_boot_secondary] cpu:1 started (0) 18
[    0.054685] [bcm2709_secondary_init] enter cpu:1
[... 323 lines cut ...]
������Rb�: x
ɠWG�'HL!Kf!P�Q0@p6D@!rQ� n(UD@!�TOSHIBA-TV
  �2UF
      �*pR !"&	�g
                                            �-��r ,%�D@!��q X,%D@!��R� �(U@D@!rQ� n(UD@!pi@raspberrypi:~ $
`

I have set up a fresh raspbian jessie SD to test this (I never use X for anything, so it's not on my system), so I have access to xrandr:

`
Screen 0: minimum 320 x 200, current 1360 x 768, maximum 2048 x 2048
HDMI-0 connected 1360x768+0+0 (normal left inverted right x axis y axis) 580mm x 320mm
   1360x768      60.02*+
   1920x1080     60.00    50.00    59.94    30.00    25.00    24.00    29.97    23.98  
   1280x1024     75.02  
   1280x720      60.00    50.00    59.94  
   1024x768      75.08    60.00  
   800x600       75.00    60.32  
   720x576       50.00  
   720x480       60.00    59.94  
   640x480       60.00    59.94  

`

Hope it helps! :)

anholt, 2016-08-15:
There's a link just below the comment entry box to attach files; that was the intent (and is required for edid).

The xrandr output there definitely shows that the 1366 hack isn't taking effect.

vanfanel, 2016-08-16:
@anholt: sorry, hadn't seen the attachment button, I am attaching the edid on this message.
So, if I understand well, the 1366 hack not taking effect is to blame for the pink column? Is there anything I can do about it?
[edid.zip](https://github.com/anholt/linux/files/419723/edid.zip)

vanfanel, 2016-08-17:
@anholt: I have built and installed today's git kernel from the raspberrypi/linux repository and the pink column is no more.

closed 2016-08-17 by vanfanel