anholt/linux #47: Does not boot with VC4 enabled (black screen)

ast, 2016-08-02:
In `/boot/config.txt` with `dtoverlay=vc4-kms-v3d` & `display_rotate=1` system does not boot. Black screen, no "rainbow" test screen. Nothing. `dtoverlay=vc4-kms-v3d` & `display_rotate=0` works as expected.

`display_rotate=1`
`gpu_mem=256`
`dtoverlay=vc4-kms-v3d`

Raspberry Pi 3.

`$ uname -a`
`Linux rpi 4.4.13-v7+ #894 SMP Mon Jun 13 13:13:27 BST 2016 armv7l GNU/Linux`

anholt, 2016-08-02:
Does display_rotate=1 work without vc4-kms-v3d?

The open driver isn't involved with the rainbow test screen, so this is something that should be reported to the Raspberry Pi foundation folks (https://github.com/raspberrypi/firmware/)

anholt, 2016-08-02:
(Also note that the open driver doesn't want any gpu_mem)

ast, 2016-08-02:
Yes, display_rotate=1 without vc4-kms-v3d works as expected.

Thanks for the information about gpu memory.

Do you mean I should cross post this issue to the the raspberry pi folks?

Not sure if there's any other information I could provide to help the debugging?

Keep up the good work! Much appreciated!

anholt, 2016-08-10:
Yeah, if a config.txt setting breaks the firmware's display (rainbow screen), then it should be reported to them first.

anholt, 2016-09-06:
Closing on our side since it's failing at the firmware level.

closed 2016-09-06 by anholt