anholt/linux #50: unclear on how to run latest kernel on rpi 3

vrodic, 2016-08-11:
This might be a documentation issue. I've tried compiling drm-vc4-next and for-next branches in order to get IF statement support in shaders on my Rpi3. I got Mesa git compiled that otherwise works on latest Rpi official kernel. 

drm-vc4-next didn't have the  Rpi3 device tree. for-next had but I had to compile the .dtb manually. drm-vc4-next seemed to be working on Rpi2, but didn't want to load mesa dri module of vc4 for some reason. drm-vc4-next booted Rpi3 to black screen when I put the rpi3 .dtb from for-next in there.

Maybe some sd card image would be useful for people who just want to test their user space apps. 

In short, how to test the latest Mesa OpenGL stack?

anholt, 2016-08-11:
Most vc4 users are just using the downstream tree.

Pi3 development happened in the last kernel, so it won't be in
drm-vc4-next yet.  Also pi3 didn't end up in the defconfig for this
kernel.

Pi3 won't do HDMI yet because of the lack of GPIO expander support.

You'd need to include some debug for the pi2 case, like
LIBGL_DEBUG=verbose.

anholt, 2016-08-30:
Closing, unless you want to follow up on some specific issue.

closed 2016-08-30 by anholt