anholt/linux #51: vc4: 7" LCD not working under X/wayland

sandersr, 2016-08-12:
Hi,

I'm trying to get Xorg and/or wayland (kwin_wayland to be precise) working under vc4. I do not have HDMI screen connected, just the 7" official DSI LCD.

I'm using the following kernel:
Linux 4.4.16-v7+ #900 SMP Wed Aug 10 18:02:31 BST 2016 armv7l armv7l armv7l GNU/Linux

and the system boots fine: [dmesg](https://gist.github.com/sandersr/e6489280800cb559b51fff11a2d9404e)

When I start sddm, it starts an Xorg server instance. [Xorg.0.log](https://gist.github.com/sandersr/fa74342b05a78176990cd76e80934ed8)

Looking at ps output, I can see Xorg running:

ps ax | grep X
  718 tty1     Ss+    0:01 /usr/libexec/Xorg -nolisten tcp -auth /var/run/sddm/{da410aef-9a1d-4af9-be98-075308f677df} -background none -noreset -displayfd 17 vt1
  728 ?        S      0:00 /bin/bash /etc/X11/xinit/Xsession /usr/bin/startkde
  738 ?        S      0:00 /bin/bash /etc/X11/xinit/Xsession /usr/bin/startkde

At this point the screen goes all black. At the same time, this line is added to the dmesg:
[Fri Aug 12 09:24:14 2016] disable

switching to another VT works fine and LCD panel greets you with a VT login prompt. Switching back to the X brings back the blank black screen.

I also manually logged in to the console and started X with just startx
The X server starts ok:
grep X
  683 tty2     S+     0:00 xinit /etc/X11/xinit/xinitrc -- /usr/bin/X :0 vt2 -keeptty -nolisten tcp -auth /home/sandersr/.serverauth.661
  684 tty2     S      0:01 /usr/libexec/Xorg :0 vt2 -keeptty -nolisten tcp -auth /home/sandersr/.serverauth.661
  686 ?        Ss     0:00 /bin/sh /etc/X11/xinit/xinitrc

but I cannot talk to it:
$ export DISPALY=:0
$ xrandr 
Can't open display 

Please let me know if you need any more information.

anholt, 2016-08-12:
You've typed DISPALY instead of DISPLAY (in case you actually copy and pasted here), but you also need your X client to get at the -auth file that the server is using.  One insecure hack is to launch X with -ac to disable auth checking (startx -- -ac I think would get that into the Xorg command line)

sandersr, 2016-08-15:
@anholt I can't believe I haven't spotted such a simple mistake with DISPLAY. Sometimes my fingers are faster than my eye...

I've got SDDM set up for user autologin so it should just start the kde session. Also running startx from the console should get the display working with X.

I _think_ the problem might be that the system believes HDMI-1 is still connected and it's trying to display everything there:

xrandr
Screen 0: minimum 320 x 200, current 800 x 600, maximum 2048 x 2048
HDMI-1 connected 800x600+0+0 (normal left inverted right x axis y axis) 0mm x 0mm
   1024x768      60.00  
   800x600       60.32* 
   640x480       59.94  
   512x384       60.00  
   400x300       60.32  
   320x240       60.05  
DSI-1 connected 800x480+0+0 (normal left inverted right x axis y axis) 0mm x 0mm
   800x480      180.15*+

When I run:
xrandr --output HDMI-1 --off

I could get a xterm running on the DSI display! I then tried running basic glxgears to see if I can get GLX working. Unfortunately I got this:

$ glxgears 
Running synchronized to the vertical refresh.  The framerate should be
approximately the same as the monitor refresh rate.
X Error of failed request:  BadAlloc (insufficient resources for operation)
  Major opcode of failed request:  148 ()
  Minor opcode of failed request:  2
  Serial number of failed request:  37
  Current serial number in output stream:  38

Running basic glxinfo also hangs. I run it with strace to see where it stops. The log is 
[here](https://gist.github.com/sandersr/effe4a71da34d63ae261b461558596b9). Not sure if that is related or shall I create a separate issue about it.

sandersr, 2016-08-15:
PS. Running sddm/kde-plasma5/glxgears/glxinfo works fine on DSI with software fallback driver.

anholt, 2016-08-15:
The HDMI force-connection is fixed in a pull request I made recently.  I don't know what would be going on with your glxinfo hang (or gears badalloc), but gdb backtrace at time of failure would be useful.

anholt, 2016-09-06:
I'm still confused by this bug report: My systems with HDMI and DSI have been displaying on both fine, and your xrandr says that HDMI and DSI are trying to scan out from the same 800x600 framebuffer, so DSI should just have the bottom 120 pixels chopped off.

sandersr, 2016-09-07:
ok, I've upgraded the kernel and I'm using the most currently available version: 4.4.19-v7+ #906 SMP Tue Aug 23. Only DSI display connected.

[dmesg](https://paste.fedoraproject.org/423250/23701914/)

Firstly, it looks like the drive doesn't pick up the display and doesn't honour these command line params: bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480
[    5.511256] vc4-drm soc:gpu: No connectors reported connected with modes
[    5.511322] [drm] Cannot find any crtc or sizes - going 1024x768

When I login to the console, the bottom pixels are still chopped off.

Now, when I start X, it looks like it's starting fine - [LOG](https://paste.fedoraproject.org/423256/47323759/) - but the display turns blank. There is no more output at all. Killing X or switching to VT makes no difference.

What's different from the precious kernel is that this line no longer show up in dmesg when starting X:
[Fri Aug 12 09:24:14 2016] disable

I will run all my tests again once mesa 12.0.2 hits fedora, which should happen at any moment now I expect.

sandersr, 2016-09-07:
I forgot to include xrandr output:

xrandr 
Screen 0: minimum 320 x 200, current 800 x 480, maximum 2048 x 2048
HDMI-1 disconnected (normal left inverted right x axis y axis)
DSI-1 connected 800x480+0+0 (normal left inverted right x axis y axis) 0mm x 0mm
   800x480      180.15*+

HDMI is no longer detected when not connected.

sandersr, 2016-09-07:
Another update.

Xorg itself starts find and can actually display stuff. The problem starts when I try to run anything GL related - but that might be down to using mesa 12.0.1. Will re-test once I've got mesa 12.0.2 installed.

sandersr, 2016-09-22:
Kernel 4.4.21-v7+ #911 and mesa 12.0.3 and everything seems to be working fine! I can start full KDE plasma 5 session and basic GL applications are working (tested on glxgears).

sandersr, 2016-09-22:
The console pixels are still chopped off tho.

anholt, 2017-01-26:
OK.  I think the console pixels chopped off are just the hpd bug from your other bug report, so closing this now.  Let me know if you have troubles on 4.9.

closed 2017-01-26 by anholt