anholt/linux #52: vc4: 7" LCD framebuffer bigger than the screen

sandersr, 2016-08-12:
Hi,

When I start RPI3 with DSI LCD screen, the framebuffer seems to be bigger than the screen. Once the cursor reaches the bottom of the screen it disappears but I can still type the commands.
[dmesg](https://gist.github.com/sandersr/e6489280800cb559b51fff11a2d9404e) for reference

anholt, 2016-08-12:
I'm guessing xrandr says something like:

Screen 0: minimum 320 x 200, current 1024 x 768, maximum 2048 x 2048
HDMI-0 connected 1024x768+0+0 (normal left inverted right x axis y axis) 527mm x 297mm
   1024x768*      60.00  
   800x600       75.00    60.32  
   640x480       75.00    60.00  
   720x400       70.08  
DSI-0 connected 800x480+0+0 (normal left inverted right x axis y axis) 0mm x 0mm
   800x480      180.15*+

If the HDMI is detected, it will be configured, and your X display with each connector at its preferred size and the X screen covering both of them.  You can change this with xrandr or your desktop environment's integrated display configuration.

The cause of this in your case would be that the downstream tree forces HDMI to connected due to some old bugs in HPD, and because we don't have a HPD solution for pi3 due to the GPIO expander not being exposed.  This needs fixing.

anholt, 2016-08-12:
(deleted wrong comment of mine)

For the console, this:

[drm] Cannot find any crtc or sizes - going 1024x768

is the problem.  It's not reconfiguring when the DSI panel shows up late, and I'm not sure offhand what the fix will be.

sandersr, 2016-08-15:
How about... Changing this:
[drm] Cannot find any crtc or sizes - going 1024x768
to the resolution of the official LCD screen? I understand it's not a proper fix, but a workaround for a current solution. Resolution of the official LCD screen is known, while 1024x768 is just a random value.

anholt, 2016-12-06:
I've realized that the hpd event is not getting sent because of a silly logic bug.  My SD card for the downstream kernel died, so I'm bringing a new one up again :(

anholt, 2017-01-26:
This seems to be working now on the 4.9 PR, so closing.  Let me know if you still have troubles on the 4.9 branch (https://github.com/raspberrypi/linux/pull/1813).

closed 2017-01-26 by anholt