anholt/linux #54: VC4: GLX under DSI displays blank window

sandersr, 2016-08-15:
Firstly it looks like vc4 doesn't work with DRI3 yet. To do anything I had to manually disable it by setting: LIBGL_DRI3_DISABLE=1

When I start X session manually and I try to run glxgears I get:

env LIBGL_DEBUG=verbose LIBGL_DRI3_DISABLE=1 glxgears
libGL: OpenDriver: trying /usr/lib/dri/tls/vc4_dri.so
libGL: OpenDriver: trying /usr/lib/dri/vc4_dri.so
libGL: Can't open configuration file /home/sandersr/.drirc: No such file or directory.
libGL: Can't open configuration file /home/sandersr/.drirc: No such file or directory.
libGL: Using DRI2 for screen 0
Attempting to import 300x300 b8g8r8x8_unorm with unsupported stride 1200 instead of 76
Running synchronized to the vertical refresh.  The framerate should be
approximately the same as the monitor refresh rate.
231 frames in 5.0 seconds = 46.066 FPS
233 frames in 5.0 seconds = 46.522 FPS
233 frames in 5.0 seconds = 46.527 FPS
233 frames in 5.0 seconds = 46.528 FPS

but the content of the window displayed on the LCD is completely blank.

env LIBGL_DEBUG=verbose LIBGL_DRI3_DISABLE=1 glxinfo | grep OpenGL
libGL: OpenDriver: trying /usr/lib/dri/tls/vc4_dri.so
libGL: OpenDriver: trying /usr/lib/dri/vc4_dri.so
libGL: Can't open configuration file /home/sandersr/.drirc: No such file or directory.
libGL: Can't open configuration file /home/sandersr/.drirc: No such file or directory.
libGL: Using DRI2 for screen 0
Attempting to import 100x100 b8g8r8x8_unorm with unsupported stride 400 instead of 28
OpenGL vendor string: Broadcom
OpenGL renderer string: Gallium 0.4 on VC4
OpenGL version string: 2.1 Mesa 12.0.1
OpenGL shading language version string: 1.20
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 2.0 Mesa 12.0.1
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 1.0.16
OpenGL ES profile extensions:

rpm -qa | grep libdrm
libdrm-2.4.70-1.fc24.armv7hl

uname -a
Linux Celica 4.4.16-v7+ #900 SMP Wed Aug 10 18:02:31 BST 2016 armv7l armv7l armv7l GNU/Linux

anholt, 2016-08-15:
DRI3 should work.  The DRI2 failure is because of:

    Attempting to import 300x300 b8g8r8x8_unorm with unsupported stride 1200 instead of 76

so probably the raspbian mesa build doesn't have 067c5b10b690b949ac28c4aea237f398f8251fff

sandersr, 2016-08-15:
I'm using fedora 24 not raspbian. Does it mean this change is not the upstream mesa yet?

anholt, 2016-08-15:
I've sent a series of commits to the mesa-stable list, so they should get merged soon.

anholt, 2016-08-15:
Also, you say this is under DSI, but I suspect it's not DSI specific.  Have you tested both ways?

sandersr, 2016-08-15:
Not sure if that helps at all, but here is there valdrind's logs for [dri2](https://gist.github.com/sandersr/645e0acfd2c86dcd2832fdacf66cccac) and [dri3](https://gist.github.com/sandersr/6a3c659a9a3b28b2e9878e10bb44d958) when running glxgears. DRI2 works, DRI3 throws:
X Error of failed request:  BadAlloc (insufficient resources for operation)

No, I haven't tried under HDMI. I'll try to find an HDMI screen and re-test it.

anholt, 2016-09-01:
DRI2: fixes are queued to stable and I've pinged the fedora folks.

gdb on _XIOError might be interesting to see what request failed.

anholt, 2016-09-06:
Most of this bug report was about the DRI2 bug, which is now fixed in the current Mesa stable release.  If you get a chance to look into the DRI3 thing, let's talk about that in a new bug report.

closed 2016-09-06 by anholt

sandersr, 2016-09-07:
Fedora should ship 12.0.2 any minute now. Once it's there, I will check DRI3 again.