anholt/linux #55: Cannot get EDID information using hdmi to vga converter.

jianhuilee, 2016-08-17:
Hi,
I am using an hdmi to vga converter on my rpi 2. If using the downstream kernel and binary blob, I can get the display's EDID by the command 'tvservice -d xxx'. But if using the upstream kernel, I get black screen because the driver cannot get display's EDID. I tried to use the command like 'i2cdump 2 0x50' and it still got nothing (kernel dumps: "i2c-bcm2835 3f805000.i2c: i2c transfer failed: 100"). I have also tried the same converter and the display on the PC to exclude the device issue. It can get the EDID by using the command 'i2cdump 2 0x50'.
Any suggenstion for debugging?
[config.txt](https://github.com/anholt/linux/files/422081/config.txt)
[dmesg.txt](https://github.com/anholt/linux/files/422082/dmesg.txt)

anholt, 2016-09-29:
If you can build your own kernel, cherry-picking commit 894200276239d2e4c60b378bdc52164fcb13af8d would be worth testing, though I don't *think* that should be your problem.

jianhuilee, 2016-09-30:
After testing that commit, the result is the same. If there is anything I can do, please tell me.
[dmesg-patch-8942002.txt](https://github.com/anholt/linux/files/502078/dmesg-patch-8942002.txt)

anholt, 2016-09-30:
OK.  We may also see improvement from https://lkml.org/lkml/2016/9/28/559 on some devices (DDC should trigger this path).

jianhuilee, 2017-05-08:
After testing upstream kernel 4.11, this issue is fixed.

closed 2017-05-08 by jianhuilee