anholt/linux #56: vc4: blank screen: can't get 1024x768 75Hz with 3D to work, works fine in 2D

LeaseOnLife, 2016-08-21:
latest firmware as of 8/21, display connected via cheap HDMI->VGA adapter

when not using 3D, it works out of the box:
tvservice -s shows:
state 0x12000a [HDMI DMT (18) RGB full 4:3], 1024x768 @ 75.00Hz, progressive
0x12000a = VC_SDTV_ATTACHED, VC_SDTV_CP_INACTIVE, **VC_HDMI_ATTACHED**, VC_HDMI_HDMI
explicitly setting the mode also works:
> /opt/vc/bin/tvservice -e "DMT 18 HDMI"
> Powering on HDMI with explicit settings (DMT mode 18)

with dtoverlay=vc4-kms-v3d, it shows the rainbow, but then shows blank screen during boot and also blank when X is started.
tvservice -s shows HDMI unplugged:
state 0x120009 [HDMI DMT (18) RGB full 4:3], 1024x768 @ 75.00Hz, progressive
0x120009 = VC_SDTV_ATTACHED, VC_SDTV_CP_INACTIVE, **VC_HDMI_UNPLUGGED**, VC_HDMI_HDMI
explicitly setting the mode fails:

>  /opt/vc/bin/tvservice -e "DMT 18 HDMI"
> Powering on HDMI with explicit settings (DMT mode 18)
> [E] Failed to power on HDMI with explicit settings (DMT mode 18)
> 

So for I have added to the /boot/config.txt
avoid_warnings=2
hdmi_drive=2
hdmi_group=2 #1=CEA 2=DMT
hdmi_mode=18
gpu_mem=128
hdmi_force_hotplug=1 # Use HDMI mode even if no HDMI monitor is detected

Anything else I can try, or should report?

Dirk

LeaseOnLife, 2016-08-21:
add/edit: PS: The older kernel from  07/28/16 works better: it also shows a black screen during boot, but X works, accelerated, glxgears running smoothly. 
tvservice -s
state 0x120009 [HDMI DMT (18) RGB full 4:3], 1024x768 @ 75.00Hz, progressive

anholt, 2016-09-22:
The open source driver has nothing to do with config.txt knobs or tvservice (tvservice should be prevented from being run with current firmware).  If you want to configure modes at boot time, take a look at https://www.kernel.org/doc/Documentation/fb/modedb.txt

I have heard we have problems with HDMI->VGA adapters -- it sounds like we lose EDID.  I do need to get one of them and test.

anholt, 2016-09-29:
Please attach (not paste) your full dmesg, and the output of xrandr --verbose while X is running.

LeaseOnLife, 2016-09-29:
will do, over the weekend!

LeaseOnLife, 2016-10-01:
attached dmesg and xrandr --verbose. with and without opensource driver (3d/2d)

[dmesg-2d.txt](https://github.com/anholt/linux/files/504448/dmesg-2d.txt)
[dmesg-3d.txt](https://github.com/anholt/linux/files/504449/dmesg-3d.txt)
[xrandr-2d.txt](https://github.com/anholt/linux/files/504450/xrandr-2d.txt)
[xrandr-3d.txt](https://github.com/anholt/linux/files/504451/xrandr-3d.txt)

anholt, 2016-10-01:
It looks like your xrandr-3d.txt got cut off.

Looking at dmesg:

    [    4.677199] vc4-drm soc:gpu: No connectors reported connected with modes
    [    4.677238] [drm] Cannot find any crtc or sizes - going 1024x768

So either the DDC didn't work or the connection detect didn't work.  Since you're on a Pi3, we don't have hotplug pin based connection detection. We fall back to a quick I2C probe against DDC, but presumably that's failing.

There's a correctness fix to the I2C for transactions like we do for DDC being settled for the upstream tree right now, but that will take a while to make it to the downstream tree.  I'd say it's fairly low likelihood of fixing this.  I also have a driver I started on to expose the hotplug pin on pi3 -- it got thoroughly bikeshedded upstream, but I should be able to merge it downstream easily.  I think that might help, as X may be saying "it's disconnected, so I won't configure that output", but I can't tell from the truncated log.

LeaseOnLife, 2016-10-01:
I double checked, what's in xrandr-3d.txt is all I get.  It does say HDMI-0 disconnected, which might cause that behavior?

```
pi@openplotter:~ $ xrandr --verbose --q12
Screen 0: minimum 320 x 200, current 1024 x 768, maximum 2048 x 2048
HDMI-0 disconnected (normal left inverted right x axis y axis)
	Identifier: 0x42
	Timestamp:  8020
	Subpixel:   unknown
	Clones:    
	CRTCs:      2
	Transform:  1.000000 0.000000 0.000000
	            0.000000 1.000000 0.000000
	            0.000000 0.000000 1.000000
	           filter: 
pi@openplotter:~ $ 
pi@openplotter:~ $ 
pi@openplotter:~ $ 
```

anholt, 2016-10-01:
OK, it looks like if the kernel doesn't attach any modes to the output (because we didn't get EDID), then X's attachment of "default" modes will not do anything.  This is maybe a thing we should fix in X.  But we should also figure out the topic of this bug: Why don't you get EDID?

LeaseOnLife, 2016-10-01:
I can only guess, edid is lost by the HDMI->VGA adapter, although the old 2D driver works.

I did try to add (actually precede) the cmdline with video=1024x768@75 which caused no harm with the 2D driver, but didn't help the 3D driver. 

I do want to mention again, with the 3D driver the boot text is lost also, before X is started, black screen after the initial splash.

LeaseOnLife, 2016-10-16:
For now I switched to the the closed source vc4-fkms-v3d, thanks for that, it seems to work.

Sekilsgs2, 2016-12-12:
raspberry 3
Hi.. So.. How fix this problem?? - i can try two monitors - black screen.
dmesg - 
[    4.677199] vc4-drm soc:gpu: No connectors reported connected with modes
[    4.677238] [drm] Cannot find any crtc or sizes - going 1024x768

what brunch from github i can use for fix this? Or any workaround?

Sekilsgs2, 2016-12-13:
Strange - i'm compile kernel 4.4-y from raspberry github  - copy new kernel, modules, overlays , dtbs to sd and screen working good. Where problem? old kernel in raspbian image?

anholt, 2017-01-30:
I bought a HDMI-VGA adapter and tried it out today on upstream 4.10-rc1 and downstream 4.9.5, and EDID has been solid on both.

@LeaseOnLife does `vcdbg log msg` show the firmware finding EDID for the monitor, or was the 1027x768@75 a mode that it defaulted to in the absence of EDID?  Does your VGA cable have all the pins present? (Some older cables didn't).  Does your VGA monitor show a correct set of modes when plugged directly into a computer with VGA out?

anholt, 2017-01-30:
Also tried going back to rpi-4.4.y to see if it was i2c changes since then that fixed things, but it still worked.

LeaseOnLife, 2017-01-30:
I don't have physical access to the Raspberry this week, I will be back
Sunday.

But I can access it via SSH, and when booted with kms (not fkms) I do
export DISPLAY=local_host:0.0 sudo vcdbg log msg
I get nothing.

My PI3 is currently at 4.9.5 and runs well using fkms, then it has boot
screen and Xwindows has the proper resolution. With kms, I see no boot
screen and the X-resulotion is set to 720x480, and more than half is off
the screen.

The Monitor supports a single resolution 1024x768m the VGA cable does have
all pins.

anholt, 2017-02-02:
Hmm, I've never seen vcdbg log message produce nothing.  I'm really not sure what would be going on there, and getting the firmware's log of what it was doing with EDID was my best hope for this bug.

LeaseOnLife, 2017-02-05:
I am back in front of my Pi3. 

> vcdbg log msg 

`
debug_sym: OpenVideoCoreMemoryFileWithOffset: Unable to open '/dev/vc-mem':
Permission denied(13)
Unable to open videocore memory access: -13
`
>sudo vcdbg log msg 
[dmesg.txt](https://github.com/anholt/linux/files/753004/dmesg.txt)

give me 217 lines of:
`debug_sym: vc_mem_copy: Failed to get memory size via ioctl: Inappropriate ioctl for device(25)`
and 
`Failed to allocate 1462923504 bytes for message buffer`

LeaseOnLife, 2017-02-05:
Sorry, didn't mean to close the issue....

anholt, 2017-02-08:
Huh, I have no idea what would be up with `sudo vcdbg log msg` there (since you're on a 32-bit kernel.  I would expect 64 to be broken).

LeaseOnLife, 2017-02-09:
With the latest "BRANCH=next rpi-upate" to kernel 4.9.8 I have the HDMI output working, during boot as well as in X, for kms and for fkms. 

Thanks!

Dirk

closed 2017-02-09 by LeaseOnLife