anholt/linux #58: Incorrect timing when using 240p 15khz (NTSC) modes

tdaede, 2016-09-12:
When using 15kHz modes (which are supported on the blob firmware via hdmi_mode=8), the actual timing differs. I've created an edid for 1920x240 (hdmi_mode=8 uses 720x240 with pixel doubling). The horizontal scan should be about 15.7kHz and vertical is about 60Hz. Actual measured timings are 12.85kHz and 50.8Hz.

The next step is probably to dump the registers and figure out where the math has gone wrong.

[1920x240.zip](https://github.com/anholt/linux/files/466353/1920x240.zip)

anholt, 2016-09-22:
Right now we don't do pixel doubling.  If you're interested in looking into it, PV_CONTROL bits 4-5 get the number of pixels repeated minus one.

anholt, 2016-09-29:
Did some hacking on this.  Fixed setting low HDMI clock rates in clk-bcm2835.c.  Fixed interlaced mode.  Fixed timeouts in transition to HDMI mode.  Pixel repeat's not quite there.  One problem is that I'm missing the HDMI infoframe, and another is that I need to be doubling the timings, not just the pixel clock.

anholt, 2016-09-29:
I've got double-clocked modes mostly scanning out now.  There's some under/overscan going on, but I don't think that's specific to this (my double-clocked  modes are all interlaced).  Code is up at https://github.com/anholt/linux/tree/drm-vc4-next-doubleclock

Now for another cup of coffee and definitely not thinking about modesetting for the rest of the day.

anholt, 2016-10-10:
Any chance you could try out that branch?

tdaede, 2016-10-10:
Yes, I got a chance to try it yesterday. I was on the 4.4.y branch before though - lots has changed since (bcm2710->bcm2835, also no 32 bit kernel for pi3). Once I get all that sorted, I'll let you know what timings I measure (including on doubled resolutions - in theory 1920x240 shouldn't be doubled because it's >25MHz)

anholt, 2017-01-30:
There have been a few more clocking fixes since then.  As far as I know this is fixed -- particularly try out the rpi-4.9.y branch if you've been using downstream kernels.  Closing, feel free to reopen if you find problems on 4.9+.

closed 2017-01-30 by anholt