anholt/linux #59: VC4 crash on 4.8rc6 when (I think) CPU idle is enabled

nullr0ute, 2016-09-19:
We were booting fine on Fedora 25 with rc6, I enabled CPU idle which we'd had previously disabled as it had caused us issues way back in the 3.x days on some ARM devices. Now it seems mostly OK but seeing this on boot on the Pi3. I think it's idle related due to debug_dma_assert_idle.

Will retest a 4.8rc7 shortly, let me know if you want more details

[   87.203288] [<c021a23c>] (unwind_backtrace) from [<c0214b94>] (show_stack+0x18/0x1c)
[   87.224256] [<c0214b94>] (show_stack) from [<c05905a0>] (dump_stack+0x9c/0xd0)
[   87.245146] [<c05905a0>] (dump_stack) from [<c0253854>] (__warn+0xe4/0x104)
[   87.265436] [<c0253854>] (__warn) from [<c02538b0>] (warn_slowpath_fmt+0x3c/0x4c)
[   87.285116] [<c02538b0>] (warn_slowpath_fmt) from [<c05be8e8>] (debug_dma_assert_idle+0x148/0x1e0)
[   87.305041] [<c05be8e8>] (debug_dma_assert_idle) from [<c03bb4a4>] (wp_page_copy+0xe8/0x63c)
[   87.325081] [<c03bb4a4>] (wp_page_copy) from [<c03bc268>] (do_wp_page+0x428/0x6f4)
[   87.345293] [<c03bc268>] (do_wp_page) from [<c03bf90c>] (handle_mm_fault+0xe00/0xed8)
[   87.365531] [<c03bf90c>] (handle_mm_fault) from [<c09ba404>] (do_page_fault+0x240/0x3b4)
[   87.385336] [<c09ba404>] (do_page_fault) from [<c0209318>] (do_DataAbort+0x40/0xc0)
[   87.404518] [<c0209318>] (do_DataAbort) from [<c09b9da0>] (__dabt_usr+0x40/0x60)
[   87.423441] Exception stack(0xd40edfb0 to 0xd40edff8)
[   87.442139] dfa0:                                     81407a58 b6e007d4 81410e40 81415d48
[   87.460488] dfc0: 81407a58 00000000 bebb07a0 b6f59c88 b6f59c88 8142cdb8 8142cae0 8142cdb8
[   87.478312] dfe0: b6e00000 bebb0780 7f62f404 7f62f0e8 a00f0010 ffffffff
[   87.496420] ---[ end trace dd231f23f09ec86f ]---
[   87.514509] Mapped at:
[   87.532475]  [<bf11a3f4>] vc4_bo_create+0x44/0xfc [vc4]
[   87.550080]  [<bf11fc20>] vc4_overflow_mem_work+0x20/0x138 [vc4]
[   87.567081]  [<c0271d94>] process_one_work+0x314/0x83c
[   87.584109]  [<c0273410>] worker_thread+0x294/0x3e4
[   87.601008]  [<c0278e44>] kthread+0xf0/0x104

nullr0ute, 2016-09-19:
that paste was a big of a fail, text version is here https://pbrobinson.fedorapeople.org/rpi3-4.8.0-0.rc6.git4.1.fc25-crash.txt

Also forgot to mention this is a minimal text install with just a text console login in HDMI

nullr0ute, 2016-09-20:
Not seeing this with the Fedora build for 4.8rc7. The difference between that and the previous build is minimal. The main difference is the first build of each kernel RC disables debug so I suspect it's something that rears it's head when debug is enabled.

anholt, 2016-09-22:
Yeah, that DMA debugging looks like it's making assertions that we're not doing things that we believe we actually can do on the platform.  (Similar to how desktop graphics drivers often violate generic DMA API assumptions, except that it's actually OK and the driver handles flushing itself).