anholt/linux #60: crash in vc4_hdmi_encoder_enable with particularl moniot

nullr0ute, 2016-09-19:
On Fedora 25 with a 4.8.0-0.rc6 kernel (with the EDID detect patch for RPi3) we're setting the following crash with a low quality monitor on a Raspberry Pi 2 (not tested on the RPi3).

The monitor is:
Haier 22 inch, hl22kn1
http://www.bradstv.com/en/catalog/product/50116-Haier-HL22KN1

EDID data:
 EDID (in hex):
   00ffffffffffff002245010001010101
   0a100103807341960acf74a3574cb023
   09484c2fce0031404540614001010101
   01010101010164190040410026301888
   360000d052000018000000fd0038551f
   3c08000a202020202020000000fc0048
   414945522054560a20202020000000fc
   000a20202020202020202020202001b6
   020317f1448405030223090707830100
   0065030c001000011d007251d01e206e
   285500c48e2100001e011d8018711c16
   20582c2500c48e2100009e8c0ad08a20
   e02d10103e9600c48e210000188c0ad0
   8a20e02d10103e9600138e2100001800
   00000000000000000000000000000000
   000000000000000000000000000000ba

[rpi2-4.8.0-0.rc6.git0.1-haier.txt](https://github.com/anholt/linux/files/481191/rpi2-4.8.0-0.rc6.git0.1-haier.txt)

anholt, 2016-09-29:
https://lists.freedesktop.org/archives/dri-devel/2016-September/119658.html should do it hopefully.

anholt, 2016-10-10:
Fix is merged to airlied's -next.

closed 2016-10-10 by anholt

nullr0ute, 2016-10-10:
For reference I pulled this into Fedora already and it'll be in the 4.8.1 kernel. When the person with the monitor is back from PTO I'll get it confirmed as fixed.