anholt/linux #64: Black screen during boot with fedora 25 beta  on HPL1925 Monitor - EDID Issue

aworan, 2016-10-19:
Monitor specs : http://h18000.www1.hp.com/products/quickspecs/archives_Division/11640_div_v4/11640_div.HTML

The result of dmesg is attached.
[dmesg.txt](https://github.com/anholt/linux/files/539750/dmesg.txt)

anholt, 2018-02-06:
Sorry for ignoring this one for so long.  Does this monitor work well on other Linux systems with the same HDMI cable?  We're getting the same EDID contents each time, so it seems like the i2c transfer is working.

aworan, 2018-02-23:
Hi Eric,
I don't have that monitor anymore, but it work with no issues with the same HDMI cable and the rapsberry pi with raspbian and the closed source driver.
It worked too with ubuntu mate for rapsberry pi and the closed source driver.

anholt, 2018-02-23:
OK, I'll have to close this down then, if we can't do tests on it.  Sorry about that!

closed 2018-02-23 by anholt