anholt/linux #65: Support for 4.8

Gerharddc, 2016-10-23:
It would be great if the VC4 driver worked on the 4.8 kernel because then we can have accelerated graphics on a 64-bit kernel (which I do actually need for my application that needs to process 64-bit numbers and draw with OpenGL ES 2).

Just a plain build of 4.8 (even 32bit) for the RPi3 (haven't tested others) doesn't seem to have to have a working VC4 setup. it compiles just fine but it fails to initialize with the following errors:

```
root@raspberrypi3:~# dmesg | grep vc4                                                                
[    1.376443] vc4-drm soc:gpu: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517                
[    1.376529] vc4-drm soc:gpu: master bind failed: -517                                             
[    4.101281] vc4-drm soc:gpu: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517                
[    4.109058] vc4-drm soc:gpu: master bind failed: -517                                             
[    4.892690] vc4-drm soc:gpu: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517                
[    4.892775] vc4-drm soc:gpu: master bind failed: -517                                             
[    4.914969] vc4-drm soc:gpu: bound 3f902000.hdmi (ops vc4_hdmi_ops)                               
[    4.915156] vc4-drm soc:gpu: bound 3f206000.pixelvalve (ops vc4_crtc_ops)                         
[    4.915292] vc4-drm soc:gpu: bound 3f207000.pixelvalve (ops vc4_crtc_ops)                         
[    4.915423] vc4-drm soc:gpu: bound 3f807000.pixelvalve (ops vc4_crtc_ops)                         
[    4.915454] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops)                                 
[    4.915501] [drm:vc4_v3d_bind] *ERROR* V3D_IDENT0 read 0xdeadbeef instead of 0x02443356           
[    4.915530] vc4-drm soc:gpu: failed to bind 3fc00000.v3d (ops vc4_v3d_ops): -22                   
[    4.915769] vc4-drm soc:gpu: master bind failed: -22                                              
[    4.915798] vc4-drm: probe of soc:gpu failed with error -22
```

At the moment I am still just trying to get HDMI up even though I will eventually actually require DSI for the official screen.

I have tried to (forward I guess)-port the VC4 code from 4.4 here to 4.8 but haven't really gotten anything to display. I do remember getting the GPU bind error to go away by trying to port the device trees from here but it still did not display anything.

Would it be possible to get it running on 4.8?

anholt, 2016-10-24:
Are you talking about downstream's 4.8 branch?  I haven't heard Dom say that it's going to be actively developed.

4.8 on pi3 upstream requires the defconfig changes and HDMI HPD workaround that are in 4.9.

Gerharddc, 2016-10-29:
I haven't tried upstream yet as I assumed downstream would be the safer bet for the pi. Are the 4.9 changes in 4.9 upstream or downstream (or both)? I don't really care about specifically 4.8 though as I just need something official/mainline/supported(ish) that can do 64bit (and VC4 because the proprietary stuff won't be an option here).

anholt, 2016-10-31:
Downstream doesn't do 64-bit at all.  You'll want 4.9-rc*.  Note that the bcm2835-defconfig-64-next patch hasn't landed yet.

anholt, 2016-11-28:
Closing this as I won't be doing any backports to 4.8.  The 4.9 kernel is likely to be the next one that downstream switches to.

closed 2016-11-28 by anholt