anholt/linux #67: HDMI audio unsupported

anholt, 2016-10-28:
The [drm-vc4-hdmi-audio](https://github.com/anholt/linux/tree/drm-vc4-hdmi-audio) branch is the work in progress for this.  Doesn't work yet.

One of the interesting problems is that the MAI_DATA input only takes in IEC60958 frames, it seems.  ALSA userspace seems to be really dependent on being able to output plain PCM samples -- everything else that does IEC958 reformats from PCM in hardware.   The firmware gets around this by reformatting PCM into IEC60958 in software, and we may need to replicate that behavior.

Electron752, 2017-01-15:
Any idea of an ETA on this driver?   The idea of having an open source HDMI audio driver rather then the closed source VCHIQ based solution sounds interesting/useful.

anholt, 2017-01-16:
Boris is hopefully taking a look this month, but the code I have is in drm-vc4-hdmi-audio if you wanted to work on debug.

Electron752, 2017-01-22:
Hi,

I was doing some digging on this and it appears that that standard requires buying.  Sounds like there may be some legal issues with a fully open source HDMI audio driver.  I don't know if it requires buying a license or what.   I did some Google searches and I found some sketchy information but nothing I read fully.  So I'm wondering if this is anything that could be included in the linux kernel either upstream or downstream.

IEC60958 appears to be a very old standard so their protection may be expired, but I'm not a lawyer.   I guess that's why most HDMI devices are closed source.

anholt, 2017-01-23:
There is no problem.  There are already many HDMI audio drivers in tree.

anholt, 2017-01-23:
Boris now has some sound coming out and is debugging the formatting of it.

Electron752, 2017-01-23:
That great, because I think this is the better way to go. But all the existing audio drivers I found in the tree are doing the formating either on a custom hardware device or in the firmware blob that gets uploaded into a controller on the hardware.

I don't think you will be able to get this checked into upstream if it violates the law for a closed standard or violates intellectual property rights.

Electron752, 2017-01-23:
And if you where planning to hook into the firmware to have it doing the firmware, I'm not sure that solution is really much better then the VCHIQ based solution.

anholt, 2017-01-23:
Format conversion is done in userspace by alsalib, and already exists for other hardware drivers.

Electron752, 2017-01-23:
OK, just so I understand.  alsalib will convert to IEC958 before sending the data to the alsa kernel driver, correct?   That way the kernel driver can pass the data to the hardware without modifying it, correct?

If I understand, this this all makes sense.  Just how would you be able to get audio to work on an old OS that hasn't updated the alsalib library yet?   Or is this completely future looking?

Electron752, 2017-01-24:
OK, I was able to get the driver to play garbage as well.   I had to compile a custom version of aplay.

One thing I noticed right away is that only supporting IEC958 breaks pulseaudio, so it makes sense to do the conversion in the driver.

What I don't understand is where to hook in the conversion.  The soc audio subsystem doesn't have a great point for adding conversion.

Electron752, 2017-01-24:
So I sent a bunch of time trying to understand this and I'm seeing alot of issues that seem to need to be resolved.

1. Why the Soc framework?  If the driver is going to do format conversion, the soc framework doesn't have a great point for adding format conversion.  It seems to me a better approach would be to make a regular ALSA driver with indirect buffering.  Kind of like how the existing ALSA driver works.

2. How would volume controls work?  If this is done in software, it's going to be hard to do it in the kernel since my understanding is that floating point isn't permitted and the volume controls are in a log scale.  Sounds like it would require lookup tables to do volume scaling, but the size of the tables would be huge.

3. The bcm2835 DMA driver can share DMA channels and IRQs between multiple clients.  Sounds like it could have some contention issues on a loaded RPI.  Since the DMA channels are essentially independent with the exception of power management(which the RPI doesn't support), perhaps it would make sense to just access the DMA channel registers directly in the audio driver rather then going through the regular DMA driver.

4. Modprobe is broken.  The VC4 driver isn't always initialized first, so the HDMI driver doesn't load.   I have to manually run modprobe on the HDMI driver from the command prompt to get it to load.

5. HDMI hotplug seems to not work.

6. How is format detection on the HDMI link and digitial copy protection going to work?

7. Finally, how are the device tree overlays going to work to disable the regular ALSA driver and what about the dependency on VC4?

One the plus side the existing VCHIQ solution seems to have broken format detection as well.  It thinks my monitor can only do 32KHz when I know for a fact it can do 44.1KHz and 48Khz.

I think the whole idea of this driver is very exciting and I'm more then interested in trying to contribute, but I'm a bit confused on the design part of it.

anholt, 2017-04-12:
This has been merged to drm-misc-next for 4.12.

closed 2017-04-12 by anholt