anholt/linux #68: Enable Hifiberry DAC with VC4 driver

joaodriessen, 2016-11-02:
This is a support related question, so I apologise for posting it here, but I could not find relevant documentation on this.

running the  4.4.28 kernel on archlinuxarm.

enabling the VC4 driver (dtoverlay=vc4-kms-v3d) I no longer get audio from my DAC (dtoverlay=hifiberry-dacplus).
Is there a way to use you video driver in conjunction with the digital analog converter from HifiBerry?

anholt, 2016-11-02:
My best bet would be that something went wrong in your config.txt setup.  I don't know of any way that vc4 should interact with non-HDMI audio.

joaodriessen, 2016-11-02:
The hardware is not being recognised by my system when I enable the video driver.

for reference, my config.txt, (maybe I screwed something up):

# See /boot/overlays/README for all available options

#hdmi_group=1
#hdmi_mode=4
#disable_overscan=1
#framebuffer_depth=32
#gpu_mem=256
dtoverlay=hifiberry-dacplus

avoid_warnings=2
dtoverlay=vc4-kms-v3d

schnitzeltony, 2016-11-02:
On Wed, Nov 2, 2016 at 9:07 PM, Joao Driessen <notifications@github.com>
wrote:

> The hardware is not being recognised by my system when I enable the video
> driver.
>
> for reference, my config.txt, (maybe I screwed something up):
> See /boot/overlays/README for all available options
>
> #hdmi_group=1
> #hdmi_mode=4
> #disable_overscan=1
> #framebuffer_depth=32
> #gpu_mem=256
> dtoverlay=hifiberry-dacplus
>
> avoid_warnings=2
> dtoverlay=vc4-kms-v3d
>
> I have same problem with a very similar config.txt:

dtoverlay=hifiberry-dacplus

# Enable VC4 Graphics
dtoverlay=vc4-kms-v3d,cma-256

My kernel is

git://github.com/raspberrypi/linux.git
branch=rpi-4.4.y
commit 1c8b82bcb72f95d8f9d606326178192a2abc9c9c (should be quite recent
4.4.27)

I attached journalctl extract - see attachment. Seems i2s wants memory
already allocated. Checked hifiberry-dacplus.dts: i2s is required by
hifiberry.

I think this is more an issue for https://github.com/raspberrypi/linux.
Shall I open an issue there?

Ah one thing - (my environment does not boot without vc4 overlay):
@*joaodriessen
<https://github.com/joaodriessen> *is that correct that hifiberry works
fine for you in case vc4 is not enabled?

joaodriessen, 2016-11-02:
Yes, as long as I don't enable Eric Anholt's driver, the hifiberry DAC+ works.

schnitzeltony, 2016-11-02:
I think this is same as https://github.com/raspberrypi/linux/issues/1420 -
w can close this one

joaodriessen, 2016-11-03:
would prefer to leave this issue open here so people who have the same issue find it and can follow the cookiecrumbs to raspberrypi/linux#1420 and raspberrypi/linux#1231

joaodriessen, 2016-11-06:
using "vc4-fkms-v3d" the HifiBerry DAC will work on kernel 4.4
But I can't get any graphics acceleration going on with this setting yet, and can't say if that's due to my system setup or that it's not ready yet, I have no idea how to properly setup this one.

anholt, 2017-01-26:
There's not much I can do for 4.4 if they're not taking the i2s clock backport.  Downstream 4.9 has cprman on by default now, so closing this.

closed 2017-01-26 by anholt