anholt/linux #69: DT: rpi3: pin for PWR led conflicts with Wi-Fi SDIO

andreamerello, 2016-11-07:
Hello,

It's look like that in both mainline and your repo (dunno if I looked at the most updated branch) in bcm2837-rpi-3-b.dts you define a PWR led attached to gpio 35, but it seems to me that this is to be assigned to Wi-Fi SDIO (alt3).

Here, on a basically mainline kernel plus sdhost, I couldn't make the Wi-Fi work until I commented out the PWR led stuff (it ended up to have that pin configured as gpio out rather than alt3). Unfortunately I don't  know where the PWR LED is attached really, so I can't craft a patch to really fix this and do a pull-request...

anholt, 2016-11-08:
So, we've been defining the pwr LED in the DT as attached to the gpio line with POWER_LOW_N input on it.  The LED is also on that line, but the firmware is sampling that line as an input continuously, so Linux shouldn't be driving that LED and we probably shouldn't be exposing it in the DT.

Regardless, on the Pi3 the POWER_LOW_N is out on the GPIO expander, which we don't have support for at all yet, so we should just remove the power LED node for now.

anholt, 2016-11-28:
a44e87b47148c6ee6b78509f47e6a15c0fae890a merged to Florian's -next.

closed 2016-11-28 by anholt