anholt/linux #71: vc4: Investigate scanning tile lists to skip loads/stores of unmodified tiles.

anholt, 2016-11-08:
Could we run a user shader in between binning completing and kicking off rendering, to scan for which tiles had no primitives binned to them and patch up the RCL to skip those?  This could massively improve X and compositor performance.  Thinking about this because I'm submitting yet another bug report for "Hey, could you scissor your rendering, please?"  Would it be worth it even if we were scanning on the CPU?

jonasarrow, 2016-11-10:
Could it work by setting the cs as the vs (no cs in this case), using a trivial rgba(1,1,1,1) fs and rendering to a downscaled render target, so that a pixel in the target corresponds to a tile in the "real" buffer. then reading the result back and only use the tiles, which are set in the buffer. Normally we would need only one tile, in extreme cases (HDR rendering or 4k displays, you would N´need some more).

A problem could be, that the rasterizer might not draw the right pixels (see https://msdn.microsoft.com/en-us/library/windows/desktop/bb147314(v=vs.85).aspx for a example of the rasterization in DX, which should be the same as OGL). If the rasterize is conforming, then we would need to pad the resulting bitmap to the right bottom about one line, which would be not optimal.

anholt, 2016-11-10:
I think the problem there would be a frame that draws (say) a single pixel is unlikely to light up a pixel center of the downscaled image.