anholt/linux #72: VC4 64-bit (f)kms not working

Gerharddc, 2016-12-07:
I am trying to get the 4.9 downstream kernel to run on VC4 with 64 bit and I am getting very close. I can now compile and run and even the fb console comes up on VC4. My ultimate goal is to use the official DSI display but at the moment I am experiencing the same situation on DSI and HDMI.

I started by porting all the patches related to DSI and when I started getting the issues I just copied the entire VC4 driver from 4.4 directly and changed the few api change related build issues to their 4.9 counterparts in order to get it to build again. This helped a bit (made plain kms work for 32-bit) but did not solve everything.

I want to get Qt5 running with EGLFS on KMS and am currently testing that and the simpler https://github.com/ds-hwang/gbm_es2_demo but they seem to behave the same.

With my 4.9 kernel as it is on https://github.com/Gerharddc/linux/tree/4.9-dsi-yocto I can get everything working on 32 bit with the kms overlay. When I use the fkms overlay though then I get the following errors from the programs after starting them:
```

using card: '/dev/dri/card0': Success
cannot find suitable CRTC for connector 24
cannot setup device for connector 0:24 (0): Success
```
`No usable crtc/encoder pair for connector "UNKNOWN1"`

On Qt this means the first frame is rendered and nothing more whilst on the other program nothing renders and the screen is just black.

The situation for fkms on 64-bit is exactly the same. On 64-bit though plain kms does not want to come up and gives the following errors:

```
[    2.369569] vc4-drm soc:gpu: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops [vc4]): -517
[    2.369673] vc4-drm soc:gpu: master bind failed: -517
[    2.392265] vc4-drm soc:gpu: failed to bind 3f902000.hdmi (ops vc4_drm_unregister [vc4]): -517
[    2.392363] vc4-drm soc:gpu: master bind failed: -517
```
I have not yet tested to see if fkms works on 4.4 (32) because I assumed it works. I will be testing it soon though just to make sure.

Assuming fkms works on 4.4, could there be something outside the VC4 codebase (because I basically ported all of it) causing fkms not to work on 4.9? Also, could there be something 64-bit related causing normal kms to fail even though vc4 compiles just fine for the architecture?

I know 64-bit isn't officially supported but if someone can give me a hint of where I should be looking to solve this issue other than the vc4 folder then I can try to figure it out myself. I guess fkms being completely broken on 4.9 is a real issue though?

PS. I am on Mesa 13.01 as on the Yocto master branch.

Gerharddc, 2016-12-08:
Ok so on the latest downstream 4.4 (32) it seems firmwarekms also doesn't work with egl/gl and also freezes after the first frame on Qt. The only error here (for dsi and hdmi) is:

```
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
```

This error was also on 4.9 but it seemed insignificant compared to the rest.

Electron752, 2017-01-16:
I was sort of able to get fkms working on 64 bit on the downstream 4.10 kernel.   I'm able to play movies with VLC with software decoding which is something I couldn't do with the basic frame buffer driver.

I do see wierd hangs in the GPU from time to time.   It seems to be somewhat dependent on the monitor resolution with hangs being more common at higher resolutions.   When the RPI hangs, it's still completely accessible through ssh, so I'm rather certain this is either a VC4 driver or GPU issue.

Gerharddc, 2017-01-16:
@Electron752 as I think I have already said, my fkms does "seem" to work on 4.10 (and from what I can tell to the same extent on 4.9) in that it brings up the graphical framebuffer and does not crash at bootup with no dmesg errors. This is not the case with plain kms though which just not want to work on 64 bit (black screen & loads of errors...).

I have not checked any VLC decoding as I do not need that. My concern is with rendering OpenGL ES 2 graphics and that does not seem to want to work. As described in https://github.com/anholt/linux/issues/78 though this seem to be a general issue with fkms on any architecture and any kernel. Hopefully once that's sorted anywhere it should work here as well.

Electron752, 2017-01-16:
Sorry, I missed that.

Plain KMS also works to some degree on 64 bit with 4.10.  The only issue is also the weird GPU resets and lockups.

I'm note sure from your post if you compiled Mesa yourself or if you got it from somewhere.   One of the requirements is that VC4 support needs to be compiled in and it isn't by default and isn't on most distros.   Debian sid arm64 does have it built-in through at this time.

Gerharddc, 2017-01-17:
@Electron752 I compiled it myself as part of a 32-bit root filesystem and it does work with plain kms and a 32-bit kernel but it does not work with fkms anywhere. I get the same behaviour when I run the Raspbian rootfs configured to use the VC4 driver through rpi-config which I guess installs the correct mesa. Can you get something like https://github.com/ds-hwang/gbm_es2_demo to run at all with a 64 bit kernel?

PS. Compliling a 64 bit rootfs and mesa does not change anything here.

Electron752, 2017-01-17:
Yep, it works fine with the downstream 4.10 kernel compiled for arm64 on debian sid arm64.

I downloaded the source, built it, ran it, and I now see the spinning colored cube on my RPI3 in ARM64 mode!

I did have to disable the X windows desktop as I get an access denied when running from X windows.  I'm guess this has something to do with it not being built on top of the X windows system and it can't run at the same time as the x windows server.

Here is what I have in my config.txt:
dtparam=audio=on
dtoverlay=vc4-kms-v3d
gpu_mem=256
mask_gpu_interrupt0=0x400   # KMS/DRM
avoid_warnings=2            # KMS/DRM
enable_uart=1
disable_overscan=1
max_usb_current=1

Here is the output of uname:
Linux pi9 4.10.0-rc3-v8+ #5 SMP PREEMPT Mon Jan 16 22:36:56 PST 2017 aarch64 GNU/Linux

Electron752, 2017-01-17:
Meshlab works quite well from the X windows desktop when running the 64 bit 4.10 kernel as well.

Here is the output of glxinfo:
name of display: :0
display: :0  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
server glx extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_profile, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_libglvnd, GLX_EXT_texture_from_pixmap, 
    GLX_EXT_visual_info, GLX_EXT_visual_rating, GLX_INTEL_swap_event, 
    GLX_MESA_copy_sub_buffer, GLX_OML_swap_method, GLX_SGIS_multisample, 
    GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group, 
    GLX_SGI_make_current_read, GLX_SGI_swap_control
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
client glx extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_profile, 
    GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_EXT_buffer_age, GLX_EXT_create_context_es2_profile, 
    GLX_EXT_create_context_es_profile, GLX_EXT_fbconfig_packed_float, 
    GLX_EXT_framebuffer_sRGB, GLX_EXT_import_context, 
    GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, GLX_EXT_visual_rating, 
    GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, 
    GLX_MESA_multithread_makecurrent, GLX_MESA_query_renderer, 
    GLX_MESA_swap_control, GLX_OML_swap_method, GLX_OML_sync_control, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control, GLX_SGI_video_sync
GLX version: 1.4
GLX extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_profile, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, 
    GLX_ARB_get_proc_address, GLX_ARB_multisample, GLX_EXT_buffer_age, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, 
    GLX_EXT_visual_rating, GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, 
    GLX_MESA_multithread_makecurrent, GLX_MESA_query_renderer, 
    GLX_MESA_swap_control, GLX_OML_swap_method, GLX_OML_sync_control, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control, GLX_SGI_video_sync
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Broadcom (0x14e4)
    Device: VC4 V3D 2.1 (0xffffffff)
    Version: 13.0.3
    Accelerated: yes
    Video memory: 724MB
    Unified memory: yes
    Preferred profile: compat (0x2)
    Max core profile version: 0.0
    Max compat profile version: 2.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 2.0
OpenGL vendor string: Broadcom
OpenGL renderer string: Gallium 0.4 on VC4 V3D 2.1
OpenGL version string: 2.1 Mesa 13.0.3
OpenGL shading language version string: 1.20
OpenGL extensions:
    GL_AMD_shader_trinary_minmax, GL_APPLE_packed_pixels, 
    GL_APPLE_vertex_array_object, GL_ARB_ES2_compatibility, 
    GL_ARB_buffer_storage, GL_ARB_clear_buffer_object, 
    GL_ARB_color_buffer_float, GL_ARB_compressed_texture_pixel_storage, 
    GL_ARB_copy_buffer, GL_ARB_debug_output, GL_ARB_depth_texture, 
[... 546 lines cut ...]
0x16f  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0  0  0  0  0  0  0  0 0 None
0x170  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0  0  0 16 16 16  0  0 0 Slow
0x171  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0  0  0  0  0  0  0  0 0 None
0x172  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0  0  0 16 16 16  0  0 0 Slow
0x173  0 dc  0  16  0 r  . .   5  6  5  0 .  .  0 24  0  0  0  0  0  0 0 None
0x174  0 dc  0  16  0 r  . .   5  6  5  0 .  .  0 24  0 16 16 16  0  0 0 Slow
0x175  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0 24  0  0  0  0  0  0 0 None
0x176  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0 24  0 16 16 16  0  0 0 Slow
0x177  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0 24  0  0  0  0  0  0 0 None
0x178  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0 24  0 16 16 16  0  0 0 Slow
0x179  0 dc  0  16  0 r  . .   5  6  5  0 .  .  0 24  8  0  0  0  0  0 0 None
0x17a  0 dc  0  16  0 r  . .   5  6  5  0 .  .  0 24  8 16 16 16  0  0 0 Slow
0x17b  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0 24  8  0  0  0  0  0 0 None
0x17c  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0 24  8 16 16 16  0  0 0 Slow
0x17d  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0 24  8  0  0  0  0  0 0 None
0x17e  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0 24  8 16 16 16  0  0 0 Slow
0x17f  0 dc  0  16  0 r  . .   5  6  5  0 .  .  0  0  0  0  0  0  0  4 1 None
0x180  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0  0  0  0  0  0  0  4 1 None
0x181  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0  0  0  0  0  0  0  4 1 None
0x182  0 dc  0  16  0 r  . .   5  6  5  0 .  .  0 24  0  0  0  0  0  4 1 None
0x183  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0 24  0  0  0  0  0  4 1 None
0x184  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0 24  0  0  0  0  0  4 1 None
0x185  0 dc  0  16  0 r  . .   5  6  5  0 .  .  0 24  8  0  0  0  0  4 1 None
0x186  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0 24  8  0  0  0  0  4 1 None
0x187  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0 24  8  0  0  0  0  4 1 None

Gerharddc, 2017-01-17:
That's great news, I'll see if I can also get it working here like that. Have you had any luck with fkms though?

Electron752, 2017-01-17:
fkms works as well.  I haven't tried it with the demo you sent me through.  I'll try it later.

I still occasionally see weird hangs in the GPU and CMA allocation issues from time to time, but I'm not sure this is anything 64 bit specific.

Electron752, 2017-01-17:
OK, I tried the demo with fkms.   And guess what... It doesn't work... Just a blank screen.

Well, at least regular kms is working.   I understand fkms is a short term solution anyway.

Gerharddc, 2017-01-17:
Yes I guess it might be a short term solution but it does not really seem to work at all. The reason I would like to have it working is to be able to have a splash screen because in normal kms the driver only kicks in way to late for this.

Electron752, 2017-01-17:
fkms works for me, and like I said I can play movies on it even though I can't with the standard frame buffer device.  It just doesn't work that well:)

As for the splash screen, the upstream kernel uses the simplefb driver to display boot info until the kernel is far enough along to load the drm based driver.   I've heard that the firmware detects somehow that the VC4 driver is going to be used, and it fakes a simplefb device in the device tree just for that purpose.  You can probably make some simple changes somewhere(probably device tree) to make it work on the downstream kernel as well.

As for simplefb, you would have to do all the graphics in software through without any hardware acceleration.  It is kind of a chicken and egg problem.

Gerharddc, 2017-01-17:
Ok I can confirm that kms now works in 4.10 64 bit even though fkms still seems to not really work anywhere. I am therefore closing this issue as there is a separate one to address fkms.

closed 2017-01-17 by Gerharddc