anholt/linux #74: ui process terminating on booting chromiumos on rpi3

giriflint, 2016-12-15:
[dmesg.txt](https://github.com/anholt/linux/files/655522/dmesg.txt)

[ui.LATEST.txt](https://github.com/anholt/linux/files/655521/ui.LATEST.txt)

[Xorg.0.log_with_hdmi.txt](https://github.com/anholt/linux/files/655520/Xorg.0.log_with_hdmi.txt)

Hi, 

I am trying to boot Chromium R53 on RaspberryPi3 using the rpi-4.4.y linux kernel which has the most recent VC4 fixes.
The system boots but without the UI. Attached are the serial and X logs, the kernel config, config.txt and the cmdline.txt. 
There was a modesetting failure as per the logs. Am I missing any config paramters in the config.txt or in the kernel .config, or, could it be a clash between two different drivers trying to access the same resource? 
Please help.

[kernel_config.txt](https://github.com/anholt/linux/files/655554/kernel_config.txt)

[cmdline.txt](https://github.com/anholt/linux/files/655548/cmdline.txt)

[config.txt](https://github.com/anholt/linux/files/655549/config.txt)

[var_log_messages.txt](https://github.com/anholt/linux/files/655525/var_log_messages.txt)

anholt, 2017-01-26:
[    0.000000] cma: Reserved 476 MiB at 0x1f000000

This cause intermittent crashes with 3D.  Delete the gpu_mem and cma lines from your config.txt, and use the defaults that vc4-kms-v3d sets up (at least, I assume you're using vc4-kms-v3d, because I don't know how the driver would probe otherwise).

The dmesg says:

    [   18.018826] vc4-drm soc:gpu: No connectors reported connected with modes
    [   18.100122] [drm] Cannot find any crtc or sizes - going 1024x768

I assume this is the error you were talking about.  This would be EDID failing to probe on HDMI (you're using HDMI, right?).  Are you by chance using an HDMI->VGA converter?

giriflint, 2017-01-27:
Thanks for the advice Eric. The error was actually in the Xorg.0.log where
the modesetting failed.

anholt, 2017-02-02:
OK, so we're looking at this:
```
[    60.934] (II) AIGLX: enabled GLX_EXT_framebuffer_sRGB
[    60.934] (II) AIGLX: enabled GLX_ARB_fbconfig_float
[    60.934] (II) AIGLX: GLX_EXT_texture_from_pixmap backed by buffer objects
[    60.938] (II) AIGLX: Loaded and initialized vc4
[    60.938] (II) GLX: Initialized DRI2 GL provider for screen 0
[    60.977] (EE) modeset(0): Failed(EE)
Fatal server error:
[    60.978] (EE) failed to create screen resources(EE)
[    60.978] (EE)
```

This is a new one to me.  Also, the mismatch between dmesg claiming no connectors with modes while Xorg.0.log shows an EDID being found is confusing.  The failure in CSR from Xorg.0.log could be from failing to set a mode, in which case dmesg from a boot with drm.debug=0x1e on the kernel command line might be useful.  But that's kind of guessing at a problem, when we would be better off using gdb to step through CreateScreenResources to see what fails.