anholt/linux #75: rpi 3 freeze when compile glmark2 from sources (when enable vc4)

Sekilsgs2, 2016-12-19:
On later raspbian, kernel 4.9 (branch=next)
i,m try compile glmark2 from sources and everytime (after reset :) ) rpi3 freeze - gui stop updating, mouse move but very very slow (2 pixel in 2 seconds). ssh stop working - cant get dmesg, sd card led - constantly on.

On firmware blobs (vc4) i'm dont have this trouble

anholt, 2016-12-19:
There's not enough information here to do anything.  Please attach at least your glmark2 build command line and how you ran the output glmark2, your config.txt, and your dmesg.

Sekilsgs2, 2016-12-20:
config.txt - default with vc4 enabled with raspi-config and gpu_mem=264.
glmark2 cloned from https://github.com/glmark2/glmark2
build instruction from readme - 

```
$ ./waf configure --with-flavors=drm-gl

To build use:

$ ./waf
```

and rpi3 freeze at 21 file. 
dmesg - cant get - rpi3 freeze and ssh stop working.

maybe problem in low ram when activate vc4 with gpu_mem=264...

Sekilsgs2, 2016-12-22:
yes - rpi take all swap memory... 
             total       used       free     shared    buffers     cached
Mem:        750148     500380     249768       5460        224      13340
-/+ buffers/cache:     486816     263332
Swap:       102396     102396          0

and freeze :)

anholt, 2016-12-27:
From the mails I got from github (I don't see the comments here, for some reason), you found that you're running out of memory.  That's not a VC4 problem, so there's nothing really to do on this bug report.

Remember, you don't need any gpu_mem setting for the vc4 driver, though, so you could at least get back a lot of your memory if you removed that.

closed 2016-12-27 by anholt