anholt/linux #76: CMA allocation issues and vc4 crash with 4.10 rc 2/3

nullr0ute, 2017-01-11:
On a vanilla 4.8.x (Fedora 25) and 4.9 kernels with mesa 13.0.x (currently 13.0.3) we use to get the occasional alloc_contig_range logged in the dmesg but there weren't particularly huge amounts of them. Now with 4.10 rc3 we're seeing a large amount of them as well as vc4 crashes.

The "alloc_contig_range: [2191e, 2191f) PFNs busy" errors were thought to be harmless but with the increased amount in 4.10 combined with the VC4 crashes I'm not so sure. The messages fill up the dmesg buffer.

nullr0ute, 2017-01-11:
Example output with Fedora workstation running gnome-shell just booted and sitting at the gdm login prompt running under Wayland:

PFNs busy lines truncated to fix where ever there's a blank line.

<pre>
[25700.021802] alloc_contig_range: [2ed00, 2ee0a) PFNs busy
[25700.021833] alloc_contig_range: [2ee00, 2ef0a) PFNs busy
[25700.021872] alloc_contig_range: [2ef00, 2f00a) PFNs busy
[25700.021906] alloc_contig_range: [2f000, 2f10a) PFNs busy
[25700.021937] alloc_contig_range: [2f100, 2f20a) PFNs busy
[25700.021967] alloc_contig_range: [2f200, 2f30a) PFNs busy
[25700.022006] alloc_contig_range: [2f300, 2f40a) PFNs busy
[25700.022041] alloc_contig_range: [2f400, 2f50a) PFNs busy
[25700.022072] alloc_contig_range: [2f500, 2f60a) PFNs busy
[25700.022102] alloc_contig_range: [2f600, 2f70a) PFNs busy
[25700.022141] alloc_contig_range: [2f700, 2f80a) PFNs busy
[25700.022180] alloc_contig_range: [2f800, 2f90a) PFNs busy
[25700.022212] alloc_contig_range: [2f900, 2fa0a) PFNs busy
[25700.022242] alloc_contig_range: [2fa00, 2fb0a) PFNs busy
[25700.022281] alloc_contig_range: [2fb00, 2fc0a) PFNs busy
[25700.022315] alloc_contig_range: [2fc00, 2fd0a) PFNs busy
[25700.022346] alloc_contig_range: [2fd00, 2fe0a) PFNs busy
[25700.022377] alloc_contig_range: [2fe00, 2ff0a) PFNs busy
[25700.022408] vc4-drm soc:gpu: failed to allocate buffer with size 1089536
[25700.023479] alloc_contig_range: [25a00, 25b0a) PFNs busy
[25700.023540] alloc_contig_range: [25b00, 25c0a) PFNs busy
[25700.023632] alloc_contig_range: [25c00, 25d0a) PFNs busy
[25700.023665] alloc_contig_range: [25d00, 25e0a) PFNs busy
[25700.023696] alloc_contig_range: [25e00, 25f0a) PFNs busy
[25700.023734] alloc_contig_range: [25f00, 2600a) PFNs busy
[25700.023769] alloc_contig_range: [26000, 2610a) PFNs busy
[25700.023804] alloc_contig_range: [26100, 2620a) PFNs busy
[25700.023834] alloc_contig_range: [26200, 2630a) PFNs busy
[25700.023873] alloc_contig_range: [26300, 2640a) PFNs busy
[25700.023907] alloc_contig_range: [26400, 2650a) PFNs busy
[25700.023938] alloc_contig_range: [26500, 2660a) PFNs busy
[25700.023968] alloc_contig_range: [26600, 2670a) PFNs busy
[25700.024007] alloc_contig_range: [26700, 2680a) PFNs busy
[25700.024041] alloc_contig_range: [26800, 2690a) PFNs busy
[25700.024072] alloc_contig_range: [26900, 26a0a) PFNs busy
[25700.024102] alloc_contig_range: [26a00, 26b0a) PFNs busy
[25700.024139] alloc_contig_range: [26b00, 26c0a) PFNs busy
[25700.024174] alloc_contig_range: [26c00, 26d0a) PFNs busy
[25700.024205] alloc_contig_range: [26d00, 26e0a) PFNs busy
[25700.024235] alloc_contig_range: [26e00, 26f0a) PFNs busy
[25700.024273] alloc_contig_range: [26f00, 2700a) PFNs busy
[25700.024307] alloc_contig_range: [27000, 2710a) PFNs busy
[25700.024338] alloc_contig_range: [27100, 2720a) PFNs busy
[25700.024368] alloc_contig_range: [27200, 2730a) PFNs busy
[25700.024407] alloc_contig_range: [27300, 2740a) PFNs busy
[25700.024441] alloc_contig_range: [27400, 2750a) PFNs busy

[25700.028399] alloc_contig_range: [2e400, 2e50a) PFNs busy
[25700.028431] alloc_contig_range: [2e500, 2e60a) PFNs busy
[25700.028461] alloc_contig_range: [2e600, 2e70a) PFNs busy
[25700.028499] alloc_contig_range: [2e700, 2e80a) PFNs busy
[25700.028533] alloc_contig_range: [2e800, 2e90a) PFNs busy
[25700.028597] alloc_contig_range: [2e900, 2ea0a) PFNs busy
[25700.028630] alloc_contig_range: [2ea00, 2eb0a) PFNs busy
[25700.028669] alloc_contig_range: [2eb00, 2ec0a) PFNs busy
[25700.028704] alloc_contig_range: [2ec00, 2ed0a) PFNs busy
[25700.028735] alloc_contig_range: [2ed00, 2ee0a) PFNs busy
[25700.028765] alloc_contig_range: [2ee00, 2ef0a) PFNs busy
[25700.028803] alloc_contig_range: [2ef00, 2f00a) PFNs busy
[25700.028837] alloc_contig_range: [2f000, 2f10a) PFNs busy
[25700.028868] alloc_contig_range: [2f100, 2f20a) PFNs busy
[25700.028898] alloc_contig_range: [2f200, 2f30a) PFNs busy
[25700.028936] alloc_contig_range: [2f300, 2f40a) PFNs busy
[25700.028971] alloc_contig_range: [2f400, 2f50a) PFNs busy
[25700.029002] alloc_contig_range: [2f500, 2f60a) PFNs busy
[... 394 lines cut ...]
[26060.021899] alloc_contig_range: [26300, 2640a) PFNs busy
[26060.021933] alloc_contig_range: [26400, 2650a) PFNs busy
[26060.021964] alloc_contig_range: [26500, 2660a) PFNs busy
[26060.021994] alloc_contig_range: [26600, 2670a) PFNs busy
[26060.022033] alloc_contig_range: [26700, 2680a) PFNs busy
[26060.022067] alloc_contig_range: [26800, 2690a) PFNs busy

[26060.027119] alloc_contig_range: [2f700, 2f80a) PFNs busy
[26060.027153] alloc_contig_range: [2f800, 2f90a) PFNs busy
[26060.027184] alloc_contig_range: [2f900, 2fa0a) PFNs busy
[26060.027221] alloc_contig_range: [2fa00, 2fb0a) PFNs busy
[26060.027296] alloc_contig_range: [2fb00, 2fc0a) PFNs busy
[26060.027334] alloc_contig_range: [2fc00, 2fd0a) PFNs busy
[26060.027366] alloc_contig_range: [2fd00, 2fe0a) PFNs busy
[26060.027396] alloc_contig_range: [2fe00, 2ff0a) PFNs busy
[26060.027425] vc4-drm soc:gpu: failed to allocate buffer with size 1089536
[26060.027538] [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
[26060.027549] [drm] num bos allocated: 173
[26060.027558] [drm] size bos allocated: 88624kb
[26060.027568] [drm] num bos used: 172
[26060.027577] [drm] size bos used: 87560kb
[26060.027586] [drm] num bos cached: 1
[26060.027595] [drm] size bos cached: 1064kb
[26060.027669] [drm:vc4_validate_bin_cl [vc4]] *ERROR* 0x00000000: packet 112 (VC4_PACKET_TILE_BINNING_MODE_CONFIG) failed to validate
</pre>

anholt, 2017-01-30:
This is fixed upstream in bc9de906a75f ("mm: alloc_contig: re-allow CMA to compact FS pages") (you should be back to 4.9 levels of dmesg spam)

closed 2017-01-30 by anholt