anholt/linux #77: DSI not working on kernel 4.9

zarr, 2017-01-11:
Hi,

Tested on RPi3 with downstream kernel (4.9.2) and 4.9.1. The DSI display is producing only blank with dtoverlay=vc4-kms-v3d. Nothing immediately suspicious in dmesg or in X.org log besides the fact that no displays are found. xranrd only reports the HDMI port.

Downstream kernel 4.4 works in X, but fails to bring up a console before that.

X.org log and dmesg output attached, what else would be helpful?

[dmesg.txt](https://github.com/anholt/linux/files/699860/dmesg.txt)
[Xorg.0.log.txt](https://github.com/anholt/linux/files/699861/Xorg.0.log.txt)

anholt, 2017-01-23:
The code hadn't been ported forward.  The PR is now sent.

anholt, 2017-02-14:
The PR has been merged.

closed 2017-02-14 by anholt