anholt/linux #79: Many GPU/VC4 Driver Hangs when using dtoverlay=vc4-kms-v3d on 4.9 RPI3 ARM64

Electron752, 2017-01-15:
Now that overlays are working in 64 bit on the downstream kernel, I tried using the vc4-kms-v3d overlay in 64 bit.   In general it works and I'm able to play YouTube videos with the Chromium web browser and play DVD quality video in VLC using software decompression.  

The problem is that I'm seeing a very large number of GPU Hangs/Resets.  The desktop will freeze sometimes for long periods of time, even though if I login to the RPI using SSH the RPI is still completely responsive and works.  This makes me think this is either a VC4 driver or GPU hardware issue.

Any ideas on what information I can provide to help debug this issue or any hints on how I can debug this issue myself?

Here is the software used:
Downstream 4.9 kernel using vc4-kms-v3d
Debian sid arm64 version
Mesa version 13 that includes VC4 support from the offical debian sid build
Chromium browser and VLC as software applications

anholt, 2017-01-31:
There's no easy debugging once you have GPU hangs.  It's all just hard guesswork.  You should probably start with attaching dmesg, though.

ktb83, 2017-01-31:
I still think some of these problems may somehow be specific to Xfce and/or related software. Using LXQT, the desktop apps/interfaces don't freeze/hang like they do on Xfce. I've used LXQT with Openbox and with Xfwm and they both seem to work OK. So maybe Xfwm/compositing isn't a part of your problem. I still have the flickering issues and poor image/text quality in general, but LXQT works much better at this point.

I could prepare an image based off my current arm64 Debian system and upload it some place if it would help save time in testing/debugging these issues.

Electron752, 2017-01-31:
I'll see if I can install LXQT locally and see if I can somehow determine what the difference is.  I probably should try installing 32 bit again and see if the same issues are present.

Thanks for the image offer, but last time I downloaded an image like that from someone my home network got hacked.  The two were probably unrelated, but you can see why caution is sometimes necessary.  Thanks for the offer though.

Electron752, 2017-02-01:
@anholt: You seem to have access to a lot of information on the GPU.  I read somewhere the GPU even has a JTAG port.  If you have access to a hardware debugger for the GPU, I suspect you could get an idea of what's wrong in a matter of a few minutes.  Where If I do it, it's going to take me alot of time since I know nothing about 3D video drivers in Linux.

Thanks to ktb83, I have the changes to get vcdbg working. I had thought that was just for the firmware. Does it apply to the GPU as well?  Is any information publicly available on how to use that tool?

ktb83, 2017-02-01:
I'd be interested to hear about what's going on with a 32-bit build. I might try building it for the Pi2B I have sitting here.

On 64-bit, the best combination of X software for me has been LXQT with Xfwm (Compositing enabled; Compton seems to exhibit various issues depending on the config used), Connman, Blueman (you can get the Bluez patches to rebuild the Debian Bluez package here https://gist.github.com/pelwell/c8230c48ea24698527cd; you may need to work those in manually, but the changes are small and few, and there are various places to get the 3 firmware files required for WiFi and Bluetooth), Pulseaudio. I played some YouTube videos in Firefox which looked pretty terrible but played steadily and the analogue audio was in sync. Audacious's OpenGL Spectrum Analyzer plug-in works pretty well and looks cool. I tried playing around with DRIconf, but none of the settings seem to make much of a difference.

Electron752, 2017-02-01:
You got firefox to work in 64 bit?  Because it crashes very easy on me.  I am able to get chromium to work fine, and yes YouTube is very slow and requires manually turning the resolution way down to make it reasonable.   I don't think it's the decoding part that's slow. The main CPUs seem to be more then adequate(I think people did software DVD decoding on a Pentium 90Mhz).  It's the GPU scaling the image and copying it to the screen that seems slow.

I didn't know Half Life was available for arm.  I wonder if Counter Strike works as well.  That was a cool game.   

I heard somewhere the 256M limit was a hardware limit, so I have no idea how 512M worked.  The CMA thing seems to certainly need improvement.

BTW, I finally got vcdbg to work last night in 64 bit.  I had to make a very minor changes to vc_mem and it turns out it's incompatible with the vc4 gpu driver.   For some strange reason, vcdbg makes a call to the dma-copy ioctl of the downstream framebuffer driver. So it's an xor kind of thing.  Either you can get some firmware debugging, or you can get open source vc4 driver.  But apparently not both!

I was hoping vcdbg allowed for real GPU debugging, but apparently it doesn't.  I was hopping it gave the ability to trace GPU instructions.  Instead it's more about poking around some variable's in the firmware that haven't had the symbols stripped. I'm not convinced the firmware even runs on the GPU.  For all I know, when the firmware driver pokes at the magic memory location, it's just triggering a self interrupt on the cpu that sends it back into the firmware.

ktb83, 2017-02-01:
That's very interesting about vcdbg, and a bummer.

closed 2017-02-03 by Electron752