anholt/linux #80: Resizing window in GeeXLab OpenGL test intermittently causes RPi3 whole system crash

cssbz, 2017-01-20:
I'm running an OpenGL test in GeeXLab, to try and get to the bottom of intermittent whole system crashes that I'm seeing.

The test shows a window containing a rotating 3D shape.

If I resize the window, the system often hangs for a few seconds.  Sometimes it crashes completely and I have to power cycle the Raspberry Pi to get things going again.  (By crashes completely, I mean it no longer responds to keyboard & mouse input, remote ssh sessions stop working, and the Raspberry Pi no longer responds to ping from another machine connected via Ethernet.)

This is what I'm using:
- Raspberry Pi 3
- I believe the PSU is OK: this is an "iClever" 6-port USB charger giving 2.4A max per port (the fourth power supply I've tried: intermittent screen flashing with the other power supplies)
- Samsung S22C200B monitor (1920 x 1080) connected via HDMI
- I believe temperature isn't an issue here: I have a heatsink installed on the Pi, and `vcgencmd measure_temp` is reporting around 56 degrees C
- I ran `sudo apt-get -y upgrade` yesterday:
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.4.38-v7+ #938 SMP Thu Dec 15 15:22:21 GMT 2016 armv7l GNU/Linux
```
- GeeXLab v0.13.0.2 from http://www.geeks3d.com/geexlab/downloads/
- Starting the OpenGL test with `./gl21.sh`

This test isn't the root of what I'm trying to fix: I'm actually trying to get full (Java) Minecraft running on the Pi for my son and I.  It works, but I'm seeing similar intermittent whole system crashes.  Oddly, I'm finding that more difficult to reproduce today (yesterday, it was crashing within a few seconds of starting Minecraft, but now playing for 15 mins doesn't reproduce it), hence trying this test.

Here are some examples from `kern.log` around the time of these crashes.

Example 1:

```
Jan 20 13:15:30 raspberrypi kernel: [  240.099813] vc4-drm soc:gpu: failed to allocate buffer with size 1069056
Jan 20 13:15:30 raspberrypi kernel: [  240.099932] vc4-drm soc:gpu: failed to allocate buffer with size 1069056
Jan 20 13:15:31 raspberrypi kernel: [  240.979931] [drm] Resetting GPU.
Jan 20 13:15:31 raspberrypi kernel: [  241.072654] [drm:vc4_validate_bin_cl [vc4]] *ERROR* 0x00000000: packet 112 (VC4_PACKET_TILE_BINNING_MODE_CONFIG) failed to validate
Jan 20 13:15:32 raspberrypi kernel: [  241.979916] [drm] Resetting GPU.
Jan 20 13:15:33 raspberrypi kernel: [  242.979960] [drm] Resetting GPU.
Jan 20 13:15:34 raspberrypi kernel: [  243.979963] [drm] Resetting GPU.
...
Jan 20 13:16:15 raspberrypi kernel: [  284.979981] [drm] Resetting GPU.
Jan 20 13:16:16 raspberrypi kernel: [  285.979989] [drm] Resetting GPU.
Jan 20 13:16:17 raspberrypi kernel: [  286.817423] Unhandled prefetch abort: section domain fault (0x009) at 0x76b6e180
```

Example 2:

```
Jan 20 13:25:35 raspberrypi kernel: [  519.988573] vc4-drm soc:gpu: failed to allocate buffer with size 1089536
Jan 20 13:25:35 raspberrypi kernel: [  519.990926] [drm:vc4_validate_bin_cl [vc4]] *ERROR* 0x00000000: packet 112 (VC4_PACKET_TILE_BINNING_MODE_CONFIG) failed to validate
Jan 20 13:25:35 raspberrypi kernel: [  519.993366] vc4-drm soc:gpu: failed to allocate buffer with size 1056768
Jan 20 13:25:35 raspberrypi kernel: [  519.993436] vc4-drm soc:gpu: failed to allocate buffer with size 1056768
Jan 20 13:25:35 raspberrypi kernel: [  520.008225] [drm:vc4_validate_bin_cl [vc4]] *ERROR* 0x00000000: packet 112 (VC4_PACKET_TILE_BINNING_MODE_CONFIG) failed to validate
Jan 20 13:25:35 raspberrypi kernel: [  520.008745] vc4-drm soc:gpu: failed to allocate buffer with size 1089536
Jan 20 13:25:35 raspberrypi kernel: [  520.008805] vc4-drm soc:gpu: failed to allocate buffer with size 1089536
Jan 20 13:25:35 raspberrypi kernel: [  520.024206] [drm:vc4_validate_bin_cl [vc4]] *ERROR* 0x00000000: packet 112 (VC4_PACKET_TILE_BINNING_MODE_CONFIG) failed to validate
Jan 20 13:25:35 raspberrypi kernel: [  520.024736] Unable to handle kernel NULL pointer dereference at virtual address 00000100
Jan 20 13:25:35 raspberrypi kernel: [  520.024747] pgd = b6dec000
Jan 20 13:25:35 raspberrypi kernel: [  520.024752] [00000100] *pgd=00000000
Jan 20 13:25:35 raspberrypi kernel: [  520.024764] Internal error: Oops: 5 [#1] SMP ARM
Jan 20 13:25:35 raspberrypi kernel: [  520.024770] Modules linked in: bnep hci_uart btbcm bluetooth brcmfmac vc4 brcmutil drm_kms_helper cfg80211 drm_mipi_dsi drm hid_apple rfkill syscopyarea sysfillrect sysimgblt snd_bcm2835 fb_sys_fops snd_pcm snd_timer snd i2c_bcm2708 bcm2835_gpiomem bcm2835_wdt evdev uio_pdrv_genirq uio i2c_dev fuse ipv6
Jan 20 13:25:35 raspberrypi kernel: [  520.024846] CPU: 3 PID: 809 Comm: Xorg Not tainted 4.4.38-v7+ #938
Jan 20 13:25:35 raspberrypi kernel: [  520.024851] Hardware name: BCM2709
Jan 20 13:25:35 raspberrypi kernel: [  520.024857] task: b8f285c0 ti: b9a2c000 task.ti: b9a2c000
Jan 20 13:25:35 raspberrypi kernel: [  520.024870] PC is at skb_copy_datagram_iter+0x100/0x250
Jan 20 13:25:35 raspberrypi kernel: [  520.024879] LR is at unix_stream_read_actor+0x30/0x3c
Jan 20 13:25:35 raspberrypi kernel: [  520.024885] pc : [<804c8184>]    lr : [<805789b0>]    psr: a0050013
Jan 20 13:25:35 raspberrypi kernel: [  520.024885] sp : b9a2dd10  ip : b9a2dd48  fp : b9a2dd44
Jan 20 13:25:35 raspberrypi kernel: [  520.024892] r10: b9a2df54  r9 : b56e09c0  r8 : 00000000
Jan 20 13:25:35 raspberrypi kernel: [  520.024898] r7 : 00000021  r6 : ffffffdf  r5 : 00000ed8  r4 : 00000100
Jan 20 13:25:35 raspberrypi kernel: [  520.024903] r3 : 00000ed8  r2 : 00000000  r1 : 00000021  r0 : b56e09c0
Jan 20 13:25:35 raspberrypi kernel: [  520.024911] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Jan 20 13:25:35 raspberrypi kernel: [  520.024917] Control: 10c5383d  Table: 36dec06a  DAC: 00000055
Jan 20 13:25:35 raspberrypi kernel: [  520.024923] Process Xorg (pid: 809, stack limit = 0xb9a2c210)
Jan 20 13:25:35 raspberrypi kernel: [  520.024929] Stack: (0xb9a2dd10 to 0xb9a2e000)
Jan 20 13:25:35 raspberrypi kernel: [  520.024936] dd00:                                     aa4c1480 b9a2de04 00000ed8 00000ed8
[... 134 lines cut ...]
Jan 20 13:25:39 raspberrypi kernel: [  523.939289] [drm] Resetting GPU.
Jan 20 13:25:41 raspberrypi kernel: [  525.939341] [drm] Resetting GPU.
Jan 20 13:25:43 raspberrypi kernel: [  527.939339] [drm] Resetting GPU.
Jan 20 13:25:45 raspberrypi kernel: [  529.939338] [drm] Resetting GPU.
Jan 20 13:25:47 raspberrypi kernel: [  531.939336] [drm] Resetting GPU.
```

Example 3:

```
Jan 20 13:46:14 raspberrypi kernel: [  246.008037] vc4-drm soc:gpu: failed to allocate buffer with size 1056768
Jan 20 13:46:14 raspberrypi kernel: [  246.012186] [drm:vc4_validate_bin_cl [vc4]] *ERROR* 0x00000000: packet 112 (VC4_PACKET_TILE_BINNING_MODE_CONFIG) failed to validate
Jan 20 13:46:14 raspberrypi kernel: [  246.012784] vc4-drm soc:gpu: failed to allocate buffer with size 1089536
Jan 20 13:46:14 raspberrypi kernel: [  246.012849] vc4-drm soc:gpu: failed to allocate buffer with size 1089536
Jan 20 13:46:15 raspberrypi kernel: [  247.038794] [drm] Resetting GPU.
Jan 20 13:46:17 raspberrypi kernel: [  249.038828] [drm] Resetting GPU.
Jan 20 13:46:19 raspberrypi kernel: [  251.038827] [drm] Resetting GPU.
...
Jan 20 13:47:01 raspberrypi kernel: [  293.038811] [drm] Resetting GPU.
Jan 20 13:47:03 raspberrypi kernel: [  294.978839] [drm] Resetting GPU.
^@^@^@^@^@^@^@^@
...
```

I hope this is the right place to report this issue (I found this bug tracker via the [New Open GL Driver, Bug report](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=135790) forum thread) - let me know if not.

cssbz, 2017-02-01:
I've just replaced the charger with a new official Raspberry Pi 3 charger.

The GeeXLab window resizing issue I describe here persists.

However, the new charger seems to have cured my intermittent whole system crashes when running (full) Minecraft, which was the problem I was actually trying to solve :)

So... feel free to close this issue if you like - it isn't important to me any more.

anholt, 2017-02-09:
I had kept this open for a bit because I was pretty sure there were crashes around CMA allocation failures, but I squashed that bug today (PR sent to 4.4.y and patch sent upstream).  I think the problem there (other than your power instability) is fixed now.

closed 2017-02-09 by anholt