anholt/linux #90: KMS/DRM: Setting CRTC on dumb buffer returns error

vanfanel, 2017-02-20:
Hi Eric,

Since I am converting some of my code from dispmanx to KMS/DRM, I have found some difficulties.
For example, I am unable to get this call working:

`drmModeSetCrtc(drm.fd, drm.crtc_id, bufs[0].fb, 0, 0, &drm.connector_id, 1, drm.mode));`

It's receiving the /dev/dri/card0 file descriptor, the current CRTC ID, a dumb buffer, a connected connector, and a drm mode struct.

The small test code where it's called is here: 
https://github.com/vanfanel/LBE_DOCS/blob/master/2D-KMS-NORMAL/context.c#L581

I suspect drmModeSetCrtc() can't be called with a dumb buffer, but according to the examples here:
https://github.com/dvdhrm/docs/blob/master/drm-howto/modeset-vsync.c
...it should be possible.
So, is it possible to call SetCrtc on a dumb buffer?

anholt, 2017-02-23:
It definitely should be possible.  I'm not at a Raspberry Pi today, but I suspect you'll find useful information in dmesg if you try this after booting with `drm.debug=0x1e` on the kernel command line.

vanfanel, 2017-02-26:
@anholt: It's possible (and drmModeSetCrtc() succeeds) if I pass it a framebuffer of the same resolution I want to set.
So, if I set a 1360x768 resolution, I have to pass drmModeSetCrtc() a 1360x768 dumb buffer. This way it works.
However, let's say I copy over a 320x200 image to that buffer. Is it possible to hardware-scale it to full screen using an overlay without having to use atomic commits to change the overlay properties?

anholt, 2017-02-26:
If you want scaling, you have to configure scaling using the atomic interfaces.  If you don't, then the DRM will correctly reject your request, since it would scan out outside of the buffer.

vanfanel, 2017-03-02:
@anholt: So, no atomic interfaces = no way to configure scaling using overlays??

anholt, 2017-03-02:
The set_plane interface might work, too.

vanfanel, 2017-03-03:
@anholt : Do you mean this function?
https://github.com/krh/weston/blob/master/src/compositor-drm.c#L594

The problem is that after calling drmModeSetCrtc(), there are no planes usable for the new CRTC. They are only available for the "original" CRTC. Any idea on why does that happen?

anholt, 2017-03-04:
I don't know what you mean by new crtc vs original crtc.  There shouldn't be any changing of CRTCs going on.

Your app starts up and there's already a mode set on the HDMI, right?  So go through the mode resources to find its CRTC, use setplane to take one of the spare planes and put it over the top of the current HDMI FB, scanning out from your smaller buffer.  From then on you could use setplane with the same crtc and plane but a new fb to present your update.  This is not as good as using the atomic interfaces, because you don't get the pageflip completion events as far as I know (and it might not be async?).

anholt, 2017-04-12:
Closing this, as there doesn't seem to be the bug described in the subject.

closed 2017-04-12 by anholt